mesh-iit/cad-libraries #12: [ECAD]-[PICOLOCK-7X1-MALE90-P150]-[11023]-[PCB Connector]

marcellosavoldi, 2020-11-05:
| |  |
|:---|:---|
| Datasheet component  | [LINK](https://www.molex.com/molex/products/part-detail/pcb_headers/5040500791) |
|maker info | [Molex - 504050-0791 ] |
| Alias Creo | [PICOLOCK-7X1-MALE90-P150] |
| MCAD Contact | [Marcello Savoldi] [ @marcellosavoldi ] |
| ECAD Contact | [Simeone Dussone] [@simeonedussoni ] |

Image with origin e PIN 1 position by PADS:

Attachment:

Notes:

### Madatory information/action:
- [ ] Added IIT-CODE
- [ ] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [ ] Added label "Libraries-Request" (could be automatic )
- [ ] Link/datasheet/drawing or other informations needing to create the libraries parts
- [ ] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)

without these informations is not possible proceed with the request issue

marcellosavoldi, 2020-11-05:
done:
[image]

simeonedussoni, 2020-11-05:
origin is coincident with PIN 1

closed 2020-11-06 by marcellosavoldi