mesh-iit/cad-libraries #13: [ECAD]-[12401598E4_USB3_1_TYPE_C]-[15047]-[PCB Connector]

marcellosavoldi, 2020-11-05:
| |  |
|:---|:---|
| Datasheet component  | [link or attachment ] |
|maker info | [Maker brand + serial number] |
| Alias Creo | [Alias or filename MCAD] |
| MCAD Contact | [Marcello Savoldi] [@marcellosavoldi ] |
| ECAD Contact | [Simeone Dussone] [@simeonedussoni ] |

Image with origin e PIN 1 position by PADS:

Attachment:

Notes:

### Madatory information/action:
- [ ] Added IIT-CODE
- [ ] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [ ] Added label "Libraries-Request" (could be automatic )
- [ ] Link/datasheet/drawing or other informations needing to create the libraries parts
- [ ] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)

without these informations is not possible proceed with the request issue

marcellosavoldi, 2020-11-05:
@simeonedussoni , any more infos about this component?

simeonedussoni, 2020-11-05:
mechanical drawings [here](https://www.amphenol-icc.com/media/wysiwyg/files/drawing/12401598xxx2a.pdf)

marcellosavoldi, 2020-11-24:
CHi @simeonedussoni ,
I presume the connector is drawed with tje PIN1 in the top-left position as the datasheet shown:
In the pre-compiled form is request the picture because the mechanical drawer needs to understand the right position of the SYS.
following my interpretation:
[image]

marcellosavoldi, 2020-11-24:
@simeonedussoni , if you confirm you can close the issue:
[image]

simeonedussoni, 2020-11-24:
correct, pin 1 is the one you guessed, can finalize and close

closed 2020-11-24 by simeonedussoni