mesh-iit/cad-libraries #14: [ECAD]-[MOLEX-UUSB-B-5-FEMSTR]-[12068]-[PCB Connector]

marcellosavoldi, 2020-11-05:
| |  |
|:---|:---|
| Datasheet component  | [LINK](https://www.molex.com/molex/products/part-detail/io_connectors/1051330011) |
|maker info | [Molex -105133-0011] |
| Alias Creo | [MOLEX-UUSB-B-5-FEMSTR] |
| MCAD Contact | [Marcello Savoldi] [@marcellosavoldi ] |
| ECAD Contact | [Simeone Dussoni] [@simeonedussoni] |

Image with origin e PIN 1 position by PADS:

Attachment:

Notes:

### Madatory information/action:
- [ ] Added IIT-CODE
- [ ] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [ ] Added label "Libraries-Request" (could be automatic )
- [ ] Link/datasheet/drawing or other informations needing to create the libraries parts
- [ ] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)

without these informations is not possible proceed with the request issue

marcellosavoldi, 2020-11-05:
@simeonedussoni , I need info about the origin of the component

simeonedussoni, 2020-11-05:
as soon as I succeed in launching the decal editor I can give you, usually is pin 1 but I need to confirm you

simeonedussoni, 2020-11-05:
Origin is centered at pin 1 
cc @marcellosavoldi

marcellosavoldi, 2020-11-24:
We change the connector with a new existing code.
see "MOLEX-UUSB-B-5-FEM90_NO-FLANGES"

closed 2020-11-24 by marcellosavoldi