mesh-iit/cad-libraries #16: [ECAD]-[AD9522-1BCPZ]-[15046]-[Programmable 12 LVDS/24 CMOS]

marcellosavoldi, 2020-11-20:
| |  |
|:---|:---|
| Datasheet component  | [link or attachment ] |
|maker info | [Maker brand + serial number] |
| Alias Creo | [AD9522-1BCPZ] |
| MCAD Contact | [Marcello Savoldi] [@marcellosavoldi ] |
| ECAD Contact | [Simeone Dussoni] [@simeonedussoni ] |

Image with origin e PIN 1 position by PADS:

Attachment:

Notes:
Component for UZCB

### Madatory information/action:
- [ ] Added IIT-CODE
- [ ] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [ ] Added label "Libraries-Request" (could be automatic )
- [ ] Link/datasheet/drawing or other informations needing to create the libraries parts
- [ ] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)

without these informations is not possible proceed with the request issue

marcellosavoldi, 2020-11-24:
@simeonedussoni , have you some more information about this component?
I noted also wingst info are missing.

marcellosavoldi, 2020-11-24:
I found this.... could be?
[image]

simeonedussoni, 2020-11-24:
ies it is

closed 2021-11-05 by simeonedussoni