mesh-iit/cad-libraries #17: ECAD-[DRV8844PWP]-[15144]-[Motor Drive]

maurizbo, 2020-11-24:
| |  |
|:---|:---|
| Datasheet component  | see Wingst |
|maker info | [ Texas Instruments - DRV8844PWP |
| Alias Creo | [Alias or filename MCAD] |
| MCAD Contact | @marcellosavoldi  |
| ECAD Contact | @maurizbo  |

Image with origin e PIN 1 position by PADS:

[image]

Attachment:

Notes:

### Madatory information/action:
- [x] Added IIT-CODE
- [x] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [x] Added label "Libraries-Request" (could be automatic )
- [x] Link/datasheet/drawing or other informations needing to create the libraries parts
- [x] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)

without these informations is not possible proceed with the request issue

fiorisi, 2021-12-15:
@maurizbo this component has been added to the library?

maurizbo, 2022-01-14:
Done, the task can be closed

closed 2022-01-14 by maurizbo