mesh-iit/cad-libraries #18: ECAD-[STM32H745IIK6]-[14464]-[Microcontroller dual core ARM Cortex-M4]

maurizbo, 2020-11-24:
| |  |
|:---|:---|
| Datasheet component  | see Wingst |
|maker info | [ST Micorelettronics - STM32H745IIK6 |
| Alias Creo | [Alias or filename MCAD] |
| MCAD Contact | @marcellosavoldi  |
| ECAD Contact | @maurizbo  |

Image with origin e PIN 1 position by PADS:

[image]

Attachment:

Notes:

### Madatory information/action:
- [x] Added IIT-CODE
- [x] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [x] Added label "Libraries-Request" (could be automatic )
- [x] Link/datasheet/drawing or other informations needing to create the libraries parts
- [x] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)

without these informations is not possible proceed with the request issue

fiorisi, 2021-12-15:
@maurizbo this component has been added to the library?

marcellosavoldi, 2021-12-17:
I can see it in the libraries...
[image]

we can close

closed 2021-12-17 by marcellosavoldi