mesh-iit/cad-libraries #19: ECAD-[TXO-2016-33-250-TR]-[15127]-[Oscillator, TCXO, 25MHz, 3.3V, 2.5ppm, SMD, 2x1.6mm]

maurizbo, 2020-11-24:
### request: 
| description| box to edit| notes|
|:---|:---|:---|
| Datasheet component  | | |
|maker info |  | |
| Alias Creo | | |
| MCAD Contact | @marcellosavoldi 
| ECAD Contact | @maurizbo | |

legend:
 "*" = To compile only if this information are missing in the database (Wingst)
  "**" = To compile only it is a preliminary name

### Image with origin e PIN 1 position by PADS 
(this info is mandatory):
[image]

###  Attachments 
(if present):
...

### Plug connector information 
( in case of connector "board to wire" is mandatory)
...

### Notes ( if needed):
...

### Madatory information/action:
(without these informations is not possible proceed with the request issue)
- [x] Added IIT-CODE
- [x] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [x] Added label "Libraries-Request" (could be automatic )
- [x] Link/datasheet/drawing or others information ( or information into Wingst database compiled)
- [x] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)
- [x] Detail about the PLUG connector (added datasheet, or Wingst code)

fiorisi, 2021-12-15:
@maurizbo this component has been added to the library?

maurizbo, 2022-01-10:
Added to the library, I close the issue

closed 2022-01-10 by maurizbo