mesh-iit/cad-libraries #21: ECAD-[MAX6741XKTGD3+T]-[15145]-[Supervisory Circuits Low-Power]

maurizbo, 2020-11-24:
### request: 
| description| box to edit| notes|
|:---|:---|:---|
| Datasheet component  | | |
|maker info | | |
| Alias Creo || |
| MCAD Contact | @marcellosavoldi  |  |
| ECAD Contact | @maurizbo  | |

legend:
 "*" = To compile only if this information are missing in the database (Wingst)
  "**" = To compile only it is a preliminary name

### Image with origin e PIN 1 position by PADS 
(this info is mandatory):
[image]

###  Attachments 
(if present):
...

### Plug connector information 
( in case of connector "board to wire" is mandatory)
...

### Notes ( if needed):
...

### Madatory information/action:
(without these informations is not possible proceed with the request issue)
- [x] Added IIT-CODE
- [x] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [x] Added label "Libraries-Request" (could be automatic )
- [x] Link/datasheet/drawing or others information ( or information into Wingst database compiled)
- [x] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)
- [x] Detail about the PLUG connector (added datasheet, or Wingst code)

fiorisi, 2021-12-15:
@maurizbo this component has been added to the library?

marcellosavoldi, 2021-12-17:
@fiorisi , I suppose no,
I noted a symbol that we cant use in Creo : 
MAX6741XKTGD3+T
I'm speaking about the plus (+).... we need to change the ALias name.

@maurizbo , where we are going to use this comp?

maurizbo, 2021-12-17:
The code MAX6741XKTGD3+T has been used of the UZCB, for the AMC rev  a and will be used in the AMC reb B

marcellosavoldi, 2021-12-21:
we absolutely have to change the Alias...

maurizbo, 2021-12-21:
Hi @marcellosavoldi I changed the alias from MAX6741XKTGD3+T to MAX6741XKTGD3PT
same I did for MAX17542GATB+T

maurizbo, 2022-01-11:
Done right now, close

closed 2022-01-11 by maurizbo