mesh-iit/cad-libraries #22: Standard Topic - [define the PLUG connector Alias name]

marcellosavoldi, 2020-11-25:
## PLUG CONNECTOR

we have to define the Alias name of all plug connector.
After this we have to move the information in the WIKI

fiorisi, 2020-11-25:
@marcellosavoldi maybe we can use the standard convention for commercial parts?
In this case would be  `manufacturer`_ `part code`.

marcellosavoldi, 2020-11-25:
We have to follow the electronics convention used until now...

fiorisi, 2020-11-25:
Ok I see. Thinking it thoroughly, in general for this kind of parts we can use directly the WINGST code, probably it is not useful to create "codici parlanti".

marcellosavoldi, 2020-11-25:
Following a list of all connector that I'go to updated every time I need to create a new plug connector, and after I'll go to move the information in the  [WIKI page](https://github.com/icub-tech-iit/cad-libraries/wiki/Plug-Connectors-mating-list)

|                  Board connector              |     Plug connector             |     ALIAS   PLUG    |     IIT-CODE     PLUG    |     Lib .part   done    |
|----------------------------------------|-----------------------------|---------------------|-------------------------|-----------------------|
|     PICOLOCK-10X1-MALE90-P150   |     MOLEX 5040511-001    |                              |                                     |                                 |
|     PICOLOCK-5X1-MALE90-P150     |     MOLEX 504051-0501    |                              |                                     |                                 |
|     PICOLOCK-4X1-MALE90-P150     |     MOLEX 504051-0401    |                              |                                     |                                 |
|     PICOLOCK-6X1-MALE90-P150     |     MOLEX 504051-0601    |                              |                                     |                                 |
|     PICOLOCK-7X1-MALE90-P150     |     MOLEX 504051-0701    |                              |                                     |                                 |
|     PICOLOCK-12X1-MALE90-P150   |     MOLEX 504051-1201    |                              |                                     |                                 |
|                                                          |                                           |                              |                                     |                                 |
|                                                          |                                           |                              |                                     |                                 |
|                                                          |                                           |                              |                                     |                                 |
|                                                          |                                           |                              |                                     |                                 |
|                                                          |                                           |                              |                                     |                                 |
|                                                          |                                           |                              |                                     |                                 |
|                                                          |                                           |                              |                                     |                                 |

marcellosavoldi, 2020-11-25:
> We have to follow the electronics convention used until now...
... I don't like an ALIAS as Wingst CODE...

fiorisi, 2020-11-25:
cc @icub-tech-iit/silo-mech

marcellosavoldi, 2020-11-25:
NEXT PLUG TO DO:
|     ALIAS BOARD CONN                              |     LIB. PART     |                            |     ALIAS PLUG CONN     |     LIB. PART    |
|-----------------------------------------------|-----------------|-------------------|----------------------------|------------------|
|     PICOBLADE-2X1-MALE90-P125              |     DONE         |     MATE WITH    |     MOLEX_51021-0200    |                  |
|     PICOBLADE-2X1-MALESTR-P125            |                        |     MATE WITH    |                         |                  |
|     PICOBLADE-3X1-MALE90-P125              |                        |     MATE WITH    |                         |                  |
|     PICOBLADE-3X1-MALESTR-P125            |                        |     MATE WITH    |                         |                  |
|     PICOBLADE-4X1-MALE90-P125              |     DONE         |     MATE WITH    |     MOLEX_51021-0400    |                  |
|     PICOBLADE-4X1-MALESTR-P125            |     DONE         |     MATE WITH    |                         |                  |
|     PICOBLADE-5X1-MALE90-P125              |     DONE         |     MATE WITH    |     MOLEX_51021-0500    |                  |
|     PICOBLADE-5X1-MALESTR-P125            |     DONE         |     MATE WITH    |                         |                  |
|     PICOBLADE-6X1-MALE90-P125              |     DONE         |     MATE WITH    |     MOLEX_51021-0600    |                  |
|     PICOBLADE-6X1-MALESTR-P125            |                        |     MATE WITH    |                         |                  |
|     PICOBLADE-7X1-MALE90-P125              |     DONE         |     MATE WITH    |     MOLEX_51021-0700    |                  |
|     PICOBLADE-7X1-MALESTR-P125            |                        |     MATE WITH    |                         |                  |
|     PICOBLADE-8X1-MALE90-P125              |     DONE         |     MATE WITH    |     MOLEX_51021-0800    |                  |
|     PICOBLADE-8X1-MALESTR-P125            |                        |     MATE WITH    |                         |                  |
|      PICOBLADE-10X1-MALE90-P125           |     DONE         |     MATE WITH    |     MOLEX_53261-1071    |                  |
|     PICOBLADE-10X1-MALESTR-P125          |     DONE         |     MATE WITH    |                         |                  |
|     PICOBLADE-12X1-MALE90-P125            |                        |     MATE WITH    |                         |                  |
|     PICOBLADE-12X1-MALESTR-P125          |     DONE         |     MATE WITH    |     MOLEX_53261-1271    |                  |
|     PICOBLADE-14X1-MALE90-P125            |                        |     MATE WITH    |                         |                  |
|                                            |                  |                  |                         |                  |

maurizbo, 2020-12-16:
NEXT PLUG TO DO:
| ALIAS BOARD CONN | LIB. PART | | ALIAS PLUG CONN | LIB. PART |
|-----------------------------------------------|-----------------|-------------------|----------------------------|------------------|
| PICOBLADE-2X1-MALE90-P125 | DONE | MATE WITH | MOLEX_51021-0200 | IIT 3002|
| PICOBLADE-2X1-MALESTR-P125 | | MATE WITH |MOLEX_51021-0200 | IIT 3002 |
| PICOBLADE-3X1-MALE90-P125 | | MATE WITH |MOLEX_51021-0300 | IIT 10716 |
| PICOBLADE-3X1-MALESTR-P125 | | MATE WITH |MOLEX_51021-0300 |IIT 10716 |
| PICOBLADE-4X1-MALE90-P125 | DONE | MATE WITH | MOLEX_51021-0400 |IIT 9 |
| PICOBLADE-4X1-MALESTR-P125 | DONE | MATE WITH |MOLEX_51021-0400 |IIT 9 |
| PICOBLADE-5X1-MALE90-P125 | DONE | MATE WITH | MOLEX_51021-0500 | IIT 702 |
| PICOBLADE-5X1-MALESTR-P125 | DONE | MATE WITH |MOLEX_51021-0500 | IIT 702 |
| PICOBLADE-6X1-MALE90-P125 | DONE | MATE WITH | MOLEX_51021-0600 | IIT 10|
| PICOBLADE-6X1-MALESTR-P125 | | MATE WITH |MOLEX_51021-0600 |IIT 10 |
| PICOBLADE-7X1-MALE90-P125 | DONE | MATE WITH | MOLEX_51021-0700 |IIT 703 |
| PICOBLADE-7X1-MALESTR-P125 | | MATE WITH |MOLEX_51021-0700 | IIT 703|
| PICOBLADE-8X1-MALE90-P125 | DONE | MATE WITH | MOLEX_51021-0800 |IIT 11 |
| PICOBLADE-8X1-MALESTR-P125 | | MATE WITH |MOLEX_51021-0800 |IIT 11 |
| PICOBLADE-10X1-MALE90-P125 | DONE | MATE WITH | MOLEX_51021-1000 |IIT 12 |
| PICOBLADE-10X1-MALESTR-P125 | DONE | MATE WITH |MOLEX_51021-1000 | IIT 12|
| PICOBLADE-12X1-MALE90-P125 | | MATE WITH |MOLEX_51021-1200 |IIT 1383 |
| PICOBLADE-12X1-MALESTR-P125 | DONE | MATE WITH | MOLEX_51021-1200|IIT 1383 |
| PICOBLADE-14X1-MALE90-P125 | | MATE WITH |MOLEX_51021-1400 |IIT 7200 |
| PICOLOCK-10X1-MALE90-P150 | |MATE WITH |MOLEX_504051-1001 | IIT 11121|
| PICOLOCK-5X1-MALE90-P150| | MATE WITH |MOLEX_504051-0501 | IIT 13076|
|PICOLOCK-4X1-MALE90-P150| | MATE WITH |MOLEX_504051-0401 | IIT  11146|
|PICOLOCK-6X1-MALE90-P150| | MATE WITH |MOLEX_504051-0601 | IIT 10878|
|PICOLOCK-7X1-MALE90-P150| | MATE WITH | MOLEX_504051-0701| IIT 11024|
|PICOLOCK-12X1-MALE90-P150| |MATE WITH | MOLEX_504051-1201| IIT 15184 |

picoblade 10X1 wrong name not 53261-1071  (in previous table) also MOLEX_53261-1271 was wrong

PICOLOCK-12X1-MALE90-P150| |MATE WITH | MOLEX_504051-1201| IIT 15184 new code

maurizbo, 2020-12-16:
@MrAndrea 
new code (plug to be used on UZCB)
PICOLOCK-12X1-MALE90-P150| |MATE WITH | MOLEX_504051-1201| IIT 15184 new code