mesh-iit/cad-libraries #27: Modification part request-[ISO7045_Z]-[added M3x14 into family table]

marcellosavoldi, 2020-11-27:
|               |                         |
|:--------------|:------------------------|
| type of part  | [MCAD ]            |
| Alias Creo    | ISO7045_Z (add V3-14--_-_ISO7045_CZ )       |
| IIT-Code      | [2542]            |
| Applicant     | Marcello Savoldi, @marcellosavoldi  |
| supervisor    | Luca Fiorio, @fiorisi  |
| other people involved |  @icub-tech-iit/silo-mech       |

## Description and motivation of the changing request: <br>
see https://github.com/icub-tech-iit/cad-libraries/issues/8

## details for the modification
added variant **V3-14--_-_ISO7045_CZ**
added a different description in the row with a different material
Could be better modify the description as Wingst
here the file excel with the right description for M1.6, M2, M2.5, M4.... we have to modify others
[new_description_iso7045_z.xlsx](https://github.com/icub-tech-iit/cad-libraries/files/5608935/new_description_iso7045_z.xlsx)

## To be filled in by mainteiners or department manager
* [ ] modification approved
* [ ] modification checked and ready for pushing to repo

fiorisi, 2021-12-15:
@marcellosavoldi can we close this issue?

marcellosavoldi, 2021-12-17:
yes... the screw has been added
[image]

closed 2021-12-17 by marcellosavoldi