mesh-iit/cad-libraries #28: ECAD -IND-2_5X2-3UH3-1A3-20TOL - 15162 - Power inductance shielded

maurizbo, 2020-11-27:
### request: 
| description| box to edit| notes|
|:---|:---|:---|
| Datasheet component  | [link or attachment ] * | |
|maker info | [Maker brand + serial number] * | |
| Alias Creo | [Alias or filename MCAD] **| |
| MCAD Contact | @marcellosavoldi  |  |
| ECAD Contact | @maurizbo  | |

legend:
 "*" = To compile only if this information are missing in the database (Wingst)
  "**" = To compile only it is a preliminary name

### Image with origin e PIN 1 position by PADS 
(this info is mandatory):
[image]

###  Attachments 
(if present):
...

### Plug connector information 
( in case of connector "board to wire" is mandatory)
...

### Notes ( if needed):
...

### Madatory information/action:
(without these informations is not possible proceed with the request issue)
- [ ] Added IIT-CODE
- [ ] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [ ] Added label "Libraries-Request" (could be automatic )
- [ ] Link/datasheet/drawing or others information ( or information into Wingst database compiled)
- [ ] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)
- [ ] Detail about the PLUG connector (added datasheet, or Wingst code)

fiorisi, 2021-12-15:
@maurizbo this component has been added to the library?

maurizbo, 2022-01-10:
Added to the library, I close the issue

closed 2022-01-10 by maurizbo