mesh-iit/cad-libraries #29: Modification part request-[ISO2338_B]-[Dowel Pins]

Elio75, 2020-11-30:
|               |                         |
|:--------------|:------------------------|
| type of part  | [MCAD ]            |
| Alias Creo    | [ISO2338_B.prt (generic)]        |
| IIT-Code      | [if present]            |
| Applicant     | @Elio75  |
| supervisor    | @marcellosavoldi  |

## Description and motivation of the changing request: <br>
We need to add tree missing variants : 
-  S1_5-12
- S1_5-14
- S1_5-16
- S1_5-20

## To be filled in by mainteiners or department manager
* [ ] modification approved
* [ ] modification checked and ready for pushing to repo

marcellosavoldi, 2020-11-30:
Following the Bossard code to add in wingst
|    |    bossard code        |   D   |  L   |
|--|-----------|-----|----|
| - | serie BN 684 - art. 1255606 | 1,5 | 12 | 
| - | serie BN 684 - art. 1117254 | 1,5 | 14 |
| - | serie BN 684 - art. 1117262 | 1,5 | 16 |
| - | serie BN 684 - art. 1117270 | 1,5 | 20 |

Elio75, 2020-12-02:
|    |    Wingst code        |   Alias  |     |
|--|-----------|-----|----|
|    |    15167        |   S1_5-12--_-_I2338_B  |     |
|    |    15168        |   S1_5-14--_-_I2338_B  |     |
|    |   15170     |   S1_5-16--_-_I2338_B  |     |
|    |   15171     |   S1_5-20--_-_I2338_B  |     |

closed 2020-12-04 by Elio75