mesh-iit/cad-libraries #33: ECAD-[DF12NC-20DS-0.5V(51)]-[ 15299]-[ Mezzanine connector, straight, SMD]

maurizbo, 2021-03-24:
### request: 
| description| box to edit| notes|
|:---|:---|:---|
| Datasheet component  | [link or attachment ] * | |
|maker info | [Maker brand + serial number] * | |
| Alias Creo | [Alias or filename MCAD] **| |
| MCAD Contact | [name Surname] [nick git] |  |
| ECAD Contact | [name Surname] [nick git] | |

legend:
 "*" = To compile only if this information are missing in the database (Wingst)
  "**" = To compile only it is a preliminary name

### Image with origin e PIN 1 position by PADS 

[image]

...

###  Attachments 
(if present):
...

### Plug connector information 

DF12NC(3.5)-20DP-0.5V(51) - DF12NC(4.0)-20DP-0.5V(51) - DF12NC(5.0)-20DP-0.5V(51)
Change height 3.5, 4.0, 5.0
### Notes ( if needed):
...

### Madatory information/action:
(without these informations is not possible proceed with the request issue)
- [x ] Added IIT-CODE
- [x ] Added assignees ( min. 1 mechanical + min. 1 Elettrical contact)
- [x ] Added label "Libraries-Request" (could be automatic )
- [x ] Link/datasheet/drawing or others information ( or information into Wingst database compiled)
- [x] The detail about the PIN 1 and the orientation of the origin of the component (img/sketch/printscreen)
- [x ] Detail about the PLUG connector (added datasheet, or Wingst code)

marcellosavoldi, 2021-08-20:
I created it...
but we have to change the ALIAS in 

> hirose-df12nc-20ds-0v5

 ....  @maurizbo , can you confirm?

[image]

maurizbo, 2021-08-23:
About the alias it was already OK

[image]

closed 2021-08-24 by marcellosavoldi