mesh-iit/cad-libraries #36: TQ_ILM50x08 & TQ_ILM50x14 - New frameless motors

divyashah, 2021-06-15:
| |  |
|:---|:---|
| Datasheet component  | [link or attachment ] |
| project | [where we are going to assembly the part libraries] |
| Alias Creo | [maker + code] |
| IIT-Code | [if present] |
| Applicant | name surname, @username |
| supervisor | name surname, @username |
| Assignees | name surname, @username |

Attachment:

Notes:

### Madatory information/action:
- [ ] Added assignees
- [ ] Added label "Libraries-Request" (could be automatic )
- [ ] Link/datasheet/drawing or other informations needing to create the libraries parts

without these informations is not possible proceed with the request issue

fiorisi, 2021-06-16:
Great @divyashah!

Can you check on the manufacturer website, or directly contact the supplier?
As seen together there are some STEP files in the ZIP attached to https://github.com/icub-tech-iit/electronics-boards/issues/76#issuecomment-788860587.

Nevertheless I would check that is the correct revision before adding it to the library.

divyashah, 2021-06-17:
> As seen together there are some STEP files in the ZIP attached to icub-tech-iit/electronics-boards#76 (comment).
>
> Nevertheless I would check that is the correct revision before adding it to the library.

It seems that the step file in the zip and also the files on the manufacturer website are the latest revision. The motors received comply with all the major dimensions.

divyashah, 2021-06-17:
> Remember also to add the [part parameters](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components), in particular follow this section for the frameless motors: https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#categories-for-frame-less-motors

Thanks @fiorisi  for pointing this out. 
Following the these guidelines and the discussion with @marcellosavoldi , we decided to design the components as follows: 
- Rotors : designed with **"features"** and assigned the type `COSM`
- Stators (with PCB) : imported from the STEP file and assigned the type `MOTR` 

The following assembly and part files will then be added to the [`cad-libraries\libraries\motors_and_gearboxes`](https://github.com/icub-tech-iit/cad-libraries/tree/master/libraries/motors_and_gearboxes):
 
```
TQ_ILM50x08.asm
├── TQ_ILM50x08_STATOR.prt
└── TQ_ILM50x08_ROTOR.prt
```

```
TQ_ILM50x14.asm
├── TQ_ILM50x14_STATOR.prt
└── TQ_ILM50x14_ROTOR.prt
```
 
> Then I think that it is also worth to align the CAD files with WINGST.

I do not have access to WINGST. Let me know how we can proceed with that part.

divyashah, 2021-06-17:
Here are some renders of the assemblies:

**TQ_ILM50x08**
[image]

**TQ_ILM50x14**
[image]

divyashah, 2021-06-17:
The following assembly and part files were committed to the [`cad-libraries\libraries\motors_and_gearboxes`](https://github.com/icub-tech-iit/cad-libraries/tree/master/libraries/motors_and_gearboxes):
 
```
TQ_ILM50x08.asm
├── TQ_ILM50x08_STATOR.prt
└── TQ_ILM50x08_ROTOR.prt
```

```
TQ_ILM50x14.asm
├── TQ_ILM50x14_STATOR.prt
└── TQ_ILM50x14_ROTOR.prt
```

Can someone review it once ?? @fiorisi  @marcellosavoldi 

cc @icub-tech-iit/silo-mech

fiorisi, 2021-06-17:
Great job @divyashah 🚀  Here some comments:

> I do not have access to WINGST. Let me know how we can proceed with that part.

I've requested the user: https://github.com/icub-tech-iit/tickets/issues/1411

> designed with "features" and assigned the type COSM

In this case please **double check** the dimensions and all the small features (e.g. chamfers). Even a small error (less than a mm) could result in a not working prototype.

Did you check the mass properties? Also for the imported part. I've added a small guide here https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts

There are some parameters that shall be removed in commercial parts. See https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components

For what concerns the description I would as to @MrAndrea to check if is fine (it is an electric motor and this description will be copied also in WINGST)
[image]

cc @icub-tech-iit/silo-mech

MrAndrea, 2021-06-18:
@fiorisi I added the `TQ` company on the _Anagrafica/Costruttori_ on Wingst:

[image]

Regarding the description, I think it is quite good but you can add something in order to have similar sentence like other motor codes.

divyashah, 2021-06-30:
Thanks @fiorisi for the feedback! 
I made updates in the parts as below: 

> In this case please **double check** the dimensions and all the small features (e.g. chamfers). Even a small error (less than a mm) could result in a not working prototype.

I didn't find a separate drawing of the rotor itself. However, I did cross check the dimensions from the assembly drawings and from the physical rotor component.  

> * Did you check the mass properties? Also for the imported part. I've added a small guide here https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts

I updated the mass properties for each part as well the assembly as per these guidelines. 

> * There are some parameters that shall be removed in commercial parts. See https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components

I removed the additional parameters as mentioned here. I kept `DESIGNED` and `DRAWN` only for the **Rotor** parts which were manually designed with "features".  Let me know if I should remove those too. 

> * This distance was taken from the datasheet?
>   [image]

Initially, I tried to place the rotor equidistant in the magnet area of the stator, so I used the distance as shown in the Fig. Now, I have updated this distance as per the datasheet. 

Below are the links to the drawing files: 
- [Drawing-ILM50x08](https://www.tq-group.com/filedownloads/files/products/robodrive/drawing/drawing_ilm50x08_rev.0102.pdf)
- [Drawing-ILM50x14](https://www.tq-group.com/filedownloads/files/products/robodrive/drawing/drawing_ilm50x14_rev.0101.pdf)

divyashah, 2021-06-30:
Thanks @MrAndrea for the feedback!

> Regarding the description, I think it is quite good but you can add something in order to have similar sentence like other motor codes.

I added the description that @marcellosavoldi suggested, which was in the description in the purchase order on `WINGST`. 

[image]

Could you maybe point me to one or two motors as example from which I can update the description ??

fiorisi, 2021-06-30:
Great @divyashah!

> I updated the mass properties for each part as well the assembly as per these guidelines.

The mass properties shall be assigned only to the parts, not to the assemblies. 

> I removed the additional parameters as mentioned here. I kept DESIGNED and DRAWN only for the Rotor parts which were manually designed with "features". Let me know if I should remove those too.

Fine for me.

> Initially, I tried to place the rotor equidistant in the magnet area of the stator, so I used the distance as shown in the Fig. Now, I have updated this distance as per the datasheet.

Great! Can you also add the datasheets to our [common folder](https://github.com/icub-tech-iit/cad-mechanics#other-material)? There is a `Motors` folder, inside feel free to create a folder for this manufacturer. In this way we can the reference it from WINGST.

divyashah, 2021-06-30:
> Great! Can you also add the datasheets to our [common folder](https://github.com/icub-tech-iit/cad-mechanics#other-material)? There is a `Motors` folder, inside feel free to create a folder for this manufacturer. In this way we can the reference it from WINGST.

I created a sub-folder named `TQ`. I added some datasheet files and also the .zip that was available from the supplier. I will keep updating it as I have more details.

fiorisi, 2021-07-02:
Great @divyashah.

I would remove the `Motor` in both the assemblies. If we use actuation for certain simulations should be located in the top level assembly. Otherwise different joint have the same `Motor`.
[image]

If I am not mistaken the weight of the ILM50x08 [should be 87 g](https://www.tq-group.com/en/products/tq-robodrive/servo-kits-brakes-and-sensors/ilm50x08/)
In the CAD assembly I see a different mass. Where did you take the weight of the rotor and stator separately? I found only the assembly weight.
[image]

fiorisi, 2021-07-02:
Also the ILM50x14 [should be 135 g](https://www.tq-group.com/en/products/tq-robodrive/servo-kits-brakes-and-sensors/ilm50x14/). While the CAD assembly seems to be 62.
[image]

divyashah, 2021-07-05:
> I would remove the `Motor` in both the assemblies. If we use actuation for certain simulations should be located in the top level assembly. Otherwise different joint have the same `Motor`.

Yes, that's true. I will remove the `Motor` from the assemblies. 

> If I am not mistaken the weight of the ILM50x08 [should be 87 g](https://www.tq-group.com/en/products/tq-robodrive/servo-kits-brakes-and-sensors/ilm50x08/)

Yes, the assembly weights for the motors (stator + rotor) for ILM50x08 and ILM50x14 are 87g and 135g respectively. 

> In the CAD assembly I see a different mass. Where did you take the weight of the rotor and stator separately? I found only the assembly weight.

I did not find the different mass specifically in datasheets. But I physically weighed the individual stator and rotor components to have the weight distribution as below: 

|     Part     |  Measured Mass [kg] | CAD Volume [mm^3] | Calculated Density [kg/mm^3] |
|:------------:|:---------:|:-------------:|:-----------------:|
| TQ_ILM50x08_Rotor  |   0.021   |    2692.87    |   7.798 e-6   |
| TQ_ILM50x08_Stator |   0.066   |    12309.5    |   5.361 e-6   |
| TQ_ILM50x14_Rotor  |   0.034   |    4385.29    |   7.753 e-6   |
| TQ_ILM50x14_Stator |   0.101   |    17872.7    |   5.651 e-6   |

---
In the CAD files, if I check the `Assigned` mass property it shows the expected mass value (~ 87g). However, the `Computed` values are different as shown in images above. 
[image]

Maybe we can cross check this together ?? 
cc @fiorisi

divyashah, 2021-07-06:
We understood that the difference in the mass properties arise due to a newer way of using mass properties in Creo 7.0. 
Here is a screenshot describing which parameters to use to assign a density to a part. 

[image]

Here are the reference links to the related guides from PTC Support for Creo 7.0: 
- [About Working with Mass Property Parameters](https://support.ptc.com/help/creo/creo_pma/r7.0/usascii/index.html#page/model-based_definition%2Fabout_working_with_mass_property_parameters.html%23) 
- [Specifying the Density of a Part](https://support.ptc.com/help/creo/creo_pma/r7.0/usascii/index.html#page/model-based_definition/specifying_the_density_of_a_part.html) 

If I remove all materials assigned (as per our guidelines on [Mass Properties of Commercial Parts](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts)), then there is no `PTC_MASS_DENSITY` parameter available. 
If I change the legacy parameter `MP_DENSITY` and assign the calculated `average density`, the mass of the Creo part is updated. 

However, there is some rounding error.  The reported mass is about **83g** where as the actual mass should be around **87g.**  

[image]

cc @fiorisi

fiorisi, 2021-07-07:
Thanks @divyashah, @salvi-mattia is experiencing the same issue. 

This could be a good case for the DEDA support team (cc @marcellosavoldi)

divyashah, 2021-07-09:
I did the following updates on the motor files: 
- Removed `Motor` from the assembly files. 
- Added a **index hole** in the Rotors that was previously missing. 
- Changed the legacy parameter `MP_DENSITY` to assign the calculated density and updated the mass properties. 

Now the computed mass from the CAD files are as follows:
- TQ_ILM50x08 : **~83g** (rated mass = 87g)
- TQ_ILM50x14 : **~142g** (rated mass = 135g)

The unit for density used in CREO is `kg/mm^3` and the assigned values are of the order of `10^-6`. Creo rounds off these density values and this results in the error in the computed mass as mentioned above. 

> This could be a good case for the DEDA support team (cc @marcellosavoldi)

So, it wold be worth asking DEDA support team if there is a better way to assign the calculated density or if we can avoid these rounding errors.

MrAndrea, 2021-07-09:
> Could you maybe point me to one or two motors as example from which I can update the description ??

Hi @divyashah, you can select the category `MO - Motors` on Wingst and for example write `frameless` in the description, then see the items:

[image]

divyashah, 2021-07-12:
Thanks @MrAndrea for the reference! 

I updated the CAD assembly files with the following description (adding rated Torque (Tr) and rated current (Ir)). 

`TQ_ILM50x08 frameless stator-rotor assembly; Tr=0.3 Nm; Ir=5.1 A`

`TQ_ILM50x14 Frameless stator-rotor assembly; Tr=0.54 Nm; Ir=5.3 A`

However, I think we still need to add these motors to the WINGST database.

fiorisi, 2021-07-14:
Great @divyashah! 🚀 

Just few questions.

> Added a index hole in the Rotors that was previously missing

Do you have a reference to document the hole location here?

> Changed the legacy parameter MP_DENSITY to assign the calculated density and updated the mass properties

Did you find this solution in the Creo help? Do you have a page to link?

Which is the description of the `_rotor` and `_stator` `prt`?

> However, I think we still need to add these motors to the WINGST database.

Yes, in this case we add only the `prt` not the assembly. @marcellosavoldi can show you how to manually create the WINGST codes or how to create a BOM to import the codes automatically.

divyashah, 2021-07-14:
Thanks @fiorisi for the feedback,

> > Added a index hole in the Rotors that was previously missing
> 
> Do you have a reference to document the hole location here?

Yes, it is given in the motor drawings. Here are the snapshots of it. 

[image]

[image]

 
> > Changed the legacy parameter MP_DENSITY to assign the calculated density and updated the mass properties
> 
> Did you find this solution in the Creo help? Do you have a page to link?

Yes, it was mentioned in the CREO help. I added the links in this comment above. ([#36 Comment](https://github.com/icub-tech-iit/cad-libraries/issues/36#issuecomment-874661606)).
> Here are the reference links to the related guides from PTC Support for Creo 7.0:
>
> * [About Working with Mass Property [...]
 
> Which is the description of the `_rotor` and `_stator` `prt`?

I kept the same description for the `_stator` &  `_rotor` as in this other comment above ([#36 Comment](https://github.com/icub-tech-iit/cad-libraries/issues/36#issuecomment-871460000)) which was also the description in the purchase order on WINGST.
> [image]

fiorisi, 2021-08-05:
@divyashah can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

divyashah, 2021-08-05:
> @divyashah can you please update the mass properties of the part accordingly to the [new [...]

Thanks @fiorisi! 
With this last commit, I updated the mass properties for the TQ motor files as per the new procedure. Now the mass shown in CREO is similar to the rated mass. 
 - TQ_ILM50x08 : ~83g~ **86.99g** (rated mass = 87g)
- TQ_ILM50x14 : ~142g~ **134.99g** (rated mass = 135g)

There is also a small update in the CAD volume and the calculated density that was reported above (due to the addition of the index hole in the rotor): 
> |     Part     |  Measured Mass [kg] | CAD Volume [mm^3] | Calculated Density [kg/mm^3] |
> |:------------:|:---------:|:-------------:|:-----------------:|
> | TQ_ILM50x08_Rotor  |   0.021   | [...]

Updating this table with correct values now: 

|     Part     |  Measured Mass [kg] | CAD Volume [mm^3] | Calculated Density [kg/mm^3] |
|:------------:|:---------:|:-------------:|:-----------------:|
| TQ_ILM50x08_Rotor  |   0.021   |    2691.69    |   7.802 e-6   |
| TQ_ILM50x08_Stator |   0.066   |    12309.5    |   5.361 e-6   |
| TQ_ILM50x14_Rotor  |   0.034   |    4384.11    |   7.755 e-6   |
| TQ_ILM50x14_Stator |   0.101   |    17872.7    |   5.651 e-6   |

fiorisi, 2021-08-05:
Great!

@marcellosavoldi do you have time to review the parts and close the issue?

fiorisi, 2021-08-11:
> @marcellosavoldi do you have time to review the parts and close the issue?

@divyashah probably you can simply add the screenshot of the mass properties as done for the other components (e.g. https://github.com/icub-tech-iit/cad-libraries/issues/45). Then we can close the issue.

divyashah, 2021-08-11:
> @divyashah probably you can simply add the screenshot of the mass properties as done for the other components (e.g. #45). Then we can close the issue.

Adding the screenshots with mass properties below: 

| **TQ_ILM50X08.ASM**  ~ **87 g** |
|:-----:| 
|[image] | 

---

| **TQ_ILM50X14.ASM**  ~ **135 g** |
|:-----:| 
| [image] |

fiorisi, 2021-08-11:
Great @divyashah. You don't have the radio buttons `Computed` or `Assigned`, correct?

divyashah, 2021-08-11:
> Great @divyashah. You don't have the radio buttons `Computed` or `Assigned`, correct?

Thanks @fiorisi!
Yes, I don't have those radio buttons anymore. The mass properties are updated as per the [new procedure in the guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts). 

| TQ_ILM50X08.ASM |
|:------:| 
| [image] | 

| TQ_ILM50X14.ASM |
|:------:| 
|  [image] |

fiorisi, 2021-11-04:
@divyashah we need to change the categories of the stator and rotor because Pamoco has 2 different codes.
Can you align with @MrAndrea for this?

MrAndrea, 2021-11-05:
Yesterday we aligned about the TQ motors, summarising: the motor is combined of two supplier codes, one for rotor and one for stator, and both must be MOT category. On Wingst they have the respectively codes as described in [this comment](https://github.com/icub-tech-iit/ergocub-design-joint/issues/9#issuecomment-961232514).

marcellosavoldi, 2021-11-05:
> @divyashah we need to change the categories of the stator and rotor because Pamoco has 2 different codes. Can you align with @MrAndrea for this?

We need to discuss how to manage these CAD codes.
We have always used two different models (Stator and Rotor), one of these has the COSM category and the other has the MOTR category ...
Is this case handled differently?
We have never written anything on WIKI, but I think it is worth talking about in the silo meeting and writing a rule.
@fiorisi , @divyashah

marcellosavoldi, 2021-11-05:
[image]

fiorisi, 2021-11-05:
> We have always used two different models (Stator and Rotor), one of these has the COSM category and the other has the MOTR category ...
Is this case handled differently?
We have never written anything on WIKI, but I think it is worth talking about in the silo meeting and writing a rule.

Definitively, we have to discuss it at the next silo meeting and update the WIKI.
For the time being I would keep the `_ROTOR` and `_STATOR`, setting the category `MOTR` for both.
In WINGST we add the correct suppliers code for the 2 parts.

divyashah, 2021-11-05:
> Yesterday we aligned about the TQ motors, summarising: the motor is combined of two supplier codes, one for rotor and one for stator, and both must be `MOTR` category. On Wingst they have the respectively codes as described in [this comment](https://github.com/icub-tech-iit/ergocub-design-joint/issues/9#issuecomment-961232514).

Thanks @MrAndrea for updating the codes on WINGST.

However, could you please cross check the suppliers codes ?? They don't seem to be correct according to me: 
[image]

I think the supplier's codes should be as per the [comment in icub-tech-iit/ergocub-design-joint#3](https://github.com/icub-tech-iit/ergocub-design-joint/issues/3#issuecomment-943258341).

> For the time being I would keep the `_ROTOR` and `_STATOR`, setting the category `MOTR` for both. 

I will update the categories to `MOTR` for all the components for now. And once we have a decision in the silo-mech meeting, I will update the WIKI accordingly. 

cc @marcellosavoldi @fiorisi

MrAndrea, 2021-11-05:
Ciao @divyashah, I didn't find the supplier codes that you [mentioned](https://github.com/icub-tech-iit/ergocub-design-joint/issues/3#issuecomment-943258341) in any document. Before changing on Wingst, where did you get them?

divyashah, 2021-11-05:
> Ciao @divyashah, I didn't find the supplier codes that you [mentioned](https://github.com/icub-tech-iit/ergocub-design-joint/issues/3#issuecomment-943258341) in any document. Before changing on Wingst, where did you get them?

@MrAndrea I found them in these drawing datasheets: [50x08](https://www.tq-group.com/filedownloads/files/products/robodrive/drawing/drawing_ilm50x08_rev.0102.pdf) and [50x14](https://www.tq-group.com/filedownloads/files/products/robodrive/drawing/drawing_ilm50x14_rev.0101.pdf). (Already added to the internal storage linked in this [comment](https://github.com/icub-tech-iit/ergocub-design-joint/issues/3#issuecomment-871541899)). 
In these datasheets they have mentioned all the various configuration codes and the ones [mentioned here](https://github.com/icub-tech-iit/ergocub-design-joint/issues/3#issuecomment-943258341) correspond to the motors we bought. I have also verified these codes with the labels on the motors we received.

MrAndrea, 2021-11-05:
Ok @divyashah they seem to me the manufacturer codes and not the supplier codes, because in the [Pamoco offer](https://github.com/icub-tech-iit/purchases/issues/1042#issuecomment-793990178) there is nothing about them.
So if you agree, I will change both the manufacturer and supplier codes with the same code provided in [your table](https://github.com/icub-tech-iit/ergocub-design-joint/issues/3#issuecomment-943258341).

divyashah, 2021-11-05:
> Ok @divyashah they seem to me the manufacturer codes and not the supplier codes, because in the [Pamoco offer](https://github.com/icub-tech-iit/purchases/issues/1042#issuecomment-793990178) there is nothing about them. 

Yes @MrAndrea you are right. They are the manufacturer codes, and not supplier codes. I got confused between the two. 

> So if you agree, I will change both the manufacturer and supplier codes with the same code provided in [your table](https://github.com/icub-tech-iit/ergocub-design-joint/issues/3#issuecomment-943258341).

Yes, then we can change both codes to the values in the table.

divyashah, 2021-11-08:
With the previous commit, I modified the rotor category to `MOTR`. 
I also added the new `IIT_CODE` parameter that was recently added to the commercial start_part based on the WINGST codes.

Once the manufacturer's codes have been updated on WINGST, we can close this issue again. 

cc @MrAndrea @fiorisi

MrAndrea, 2021-11-08:
Manufacturer's codes updated on Wingst.

[image]

[image]

[image]

[image]

fiorisi, 2021-11-11:
@MrAndrea @divyashah I think that we can close, right?

marcellosavoldi, 2021-11-11:
Let's remember to discuss it at the next Silo Mech meeting ...
I think it is useful to reiterate that there are exceptions to entering parameters into the model.
cc: @fiorisi , @icub-tech-iit/silo-mech

closed 2021-11-11 by divyashah