mesh-iit/cad-libraries #39: [ICYBOX_IB-AC527] - [USB Sound Card Adapter]

salvi-mattia, 2021-07-05:
| Description | Field |
|:---|:---|
| Datasheet component  | https://www.distrelec.it/Web/Downloads/_t/ds/ib-ac527_eng_tds.pdf |
| project | icub head |
| Alias Creo | ICYBOX_IB-AC527 |
| IIT-Code | 5988  |
| Applicant | @salvi-mattia  |
| supervisor | @fiorisi  |

### Attachment:

### Notes:

### Checklist:
- [x] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [x] File imported using the Creo [start part](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#cad-documentation-standards).
- [x] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [x] Edited the [parameters](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [x] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

#### NOTE
Without these informations is not possible proceed with the request.

maggia80, 2021-07-05:
@salvi-mattia this component was used in the talking-head we should have already the part

salvi-mattia, 2021-07-05:
maybe it's similar

[image]

I created this one
https://www.distrelec.it/Web/Downloads/_t/ds/ib-ac527_eng_tds.pdf

salvi-mattia, 2021-07-05:
according to @pattacini I'm going to use the already existing component reported here above by @maggia80 

I'm not going to commit the new icybox component (cc @fiorisi )

fiorisi, 2021-07-06:
> according to @pattacini I'm going to use the already existing component reported here above by @maggia80

I'm not going to commit the new icybox component (cc @fiorisi )

salvi-mattia, 2021-07-07:
> 
> 
> > according to @pattacini I'm going to use the already existing component reported here above by @maggia80
> > I'm not going to commit the new icybox component (cc @fiorisi )
> 
> Great, meanwhile can you please check that the component satisfies the [checklist](https://github.com/icub-tech-iit/cad-libraries/issues/39#issue-937119083)?

DONE

fiorisi, 2021-07-07:
@salvi-mattia can you check the same issues commenter here https://github.com/icub-tech-iit/cad-libraries/issues/40#issuecomment-875448941?

salvi-mattia, 2021-07-07:
it's not possible to assign a specifc density to the component due to the problem of materials and bodies seen with @fiorisi 
in the meanwhile I'm going to assign it an existing material with similar mass properties

fiorisi, 2021-08-05:
@salvi-mattia can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

salvi-mattia, 2021-08-06:
resolved with eccbe7a60caf917aec284b9cdf9f08fb6fe4cd39

@fiorisi i think we can close, do you agree?

fiorisi, 2021-08-09:
> @fiorisi i think we can close, do you agree?

Can you just add a screenshot of the mass properties?

salvi-mattia, 2021-08-09:
[image]

closed 2021-08-09 by salvi-mattia