mesh-iit/cad-libraries #40: SWITCHCRAFT_35RASMT4BHNTRX - 	PCB connector, jack stereo, 5 contact

salvi-mattia, 2021-07-07:
| Description | Field |
|:---|:---|
| Datasheet component  | [link or attachment ] |
| project | iCub head |
| Alias Creo |SWITCHCRAFT_35RASMT4BHNTRX|
| IIT-Code | 5280  |
| Applicant | @salvi-mattia  |
| supervisor | @fiorisi  |

### Attachment:

### Notes:

### Checklist:
- [x] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [x] File imported using the Creo [start part](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#cad-documentation-standards).
- [x] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [x] Edited the [parameters](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

#### NOTE
Without these informations is not possible proceed with the request.

[image]

[image]

fiorisi, 2021-07-07:
Thanks @salvi-mattia 🚀 

Just few comments.

The material assigned is 17-4PH. Please follow the [guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts) 😉 
[image]

Some parameters in commercial parts [should be removed](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components). Probably we can re-discuss if we want to keep the `DESIGNED` and `DRAWN` for part that have custom features (i.e. not imported). -> added a point in the [silo-mech agenda](https://istitutoitalianotecnologia.sharepoint.com/sites/icub-facility/_layouts/OneNote.aspx?id=%2Fsites%2Ficub-facility%2FDocumenti%20condivisi%2FSilo%20Mechanics%2Fchat_attachments%2FMeeting%20Notes&wd=target%28MoM.one%7CEB34C784-605E-40F6-9CDC-19755C5821A4%2FSilo%20Meeting%20-%2015%20July%202021%7C804FEACA-EC50-458F-94F0-33A9E84E2DC5%2F%29
onenote:https://istitutoitalianotecnologia.sharepoint.com/sites/icub-facility/Documenti%20condivisi/Silo%20Mechanics/chat_attachments/Meeting%20Notes/MoM.one#Silo%20Meeting%20-%2015%20July%202021&section-id={EB34C784-605E-40F6-9CDC-19755C5821A4}&page-id={804FEACA-EC50-458F-94F0-33A9E84E2DC5}&end).
[image]

salvi-mattia, 2021-07-07:
as we said on teams, this is a steel part both in real life and in cad, 
it weights exactly the same both in cad and in RL
the question is: do I have to change it again?

salvi-mattia, 2021-07-07:
it's not possible to assign a specifc density to the component due to the problem of materials and bodies seen with @fiorisi 
in the meanwhile I'm going to assign it an existing material with similar mass properties

fiorisi, 2021-08-05:
@salvi-mattia can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

salvi-mattia, 2021-08-06:
resolved with eccbe7a60caf917aec284b9cdf9f08fb6fe4cd39

@fiorisi i think we can close, do you agree?

salvi-mattia, 2021-08-09:
[image]

closed 2021-08-09 by salvi-mattia