mesh-iit/cad-libraries #41: PWB_ENCODER-103839 - AMO magnetic target

salvi-mattia, 2021-07-07:
| Description | Field |
|:---|:---|
| Datasheet component  | file:///C:/Users/msalvi/AppData/Local/Temp/MU7S_64-63N_PWB.pdf|
| project | [where we are going to assembly the part libraries] |
| Alias Creo | PWB_ENCODER-103839  |
| IIT-Code | 10077 |
| Applicant | @salvi-mattia  |
| supervisor | @fiorisi  |

### Attachment:

### Notes:
it's not possible to assign a specifc density to the component due to the problem of materials and bodies seen with @fiorisi 
in the meanwhile I'm going to assign it an existing material with similar mass properties

### Checklist:
- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#cad-documentation-standards).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Edited the [parameters](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

#### NOTE
Without these informations is not possible proceed with the request.

fiorisi, 2021-08-05:
@salvi-mattia can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

salvi-mattia, 2021-08-06:
resolved with eccbe7a60caf917aec284b9cdf9f08fb6fe4cd39

@fiorisi i think we can close, do you agree?

salvi-mattia, 2021-08-09:
[image]

marcellosavoldi, 2021-08-12:
I'm sorry @salvi-mattia , can I checked the box as done?
[image]

closed 2021-08-13 by salvi-mattia