mesh-iit/cad-libraries #42: [manufacturer] - [part number] - [short description]

Mick3Lozzo, 2021-07-08:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  |  [Intel-RealSense-D400-Series-Datasheet-June-2020.pdf](https://github.com/icub-tech-iit/cad-libraries/files/6784597/Intel-RealSense-D400-Series-Datasheet-June-2020.pdf) |
| Project              | ergoCub  |
| Alias Creo           | INTEL_REALSENSE_D450   |
| IIT-Code             | XXXXX                        |
| Applicant            | Michele Gesino, @Mick3Lozzo    |
| Supervisor           | Ugo Pattacini, @pattacini   |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#cad-documentation-standards).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Edited the [parameters](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

Mick3Lozzo, 2021-07-08:
*   The item still does not have an IIT-Code
*   Mass properties are uncertain, we have to check when we buy one.

fiorisi, 2021-07-08:
@Mick3Lozzo can you please edit the issue title?

maggia80, 2021-07-12:
[image]

and bottom:

[image]

Mick3Lozzo, 2021-07-13:
@maggia80 I've found the CAD model of the board. Does it already have a WinGST code?

[image]

maggia80, 2021-07-13:
to connect the module with the board there are two options:

1. from Intel: a rigid PCB that fix the two parts one over the other more or less like in the picture above. Mouser link https://www.mouser.it/ProductDetail/Intel/82635DSITR50P?qs=wd5RIQLrsJhIJqpBcxiaKQ%3D%3D

2. a flexible PCB with different length provided by FRAMOS: https://www.framos.com/en/hsf50-22273 This option is preferable we should ask for a quotation or if FRAMOS could add these codes in the Mouser catalog. There are already products from FRAMOS available at Mouser. 

There aren't wingst codes for modules and cables

fiorisi, 2021-07-14:
Great @Mick3Lozzo!

Is it possible to remove all the small features from the [3D model](https://github.com/icub-tech-iit/cad-libraries/issues/42#issuecomment-878903609) so that is lighter to open and render?

Mick3Lozzo, 2021-07-14:
**D450 weight**: 29,8 g

Mick3Lozzo, 2021-07-14:
> Is it possible to remove all the small features from the [3D model](https://github.com/icub-tech-iit/cad-libraries/issues/42#issuecomment-878903609) so that is lighter to open and render?

Probably. I imported directly the step file as it was, we could re-import it or work on the already imported part.

Mick3Lozzo, 2021-07-14:
As already reported by @divyashah [here](https://github.com/icub-tech-iit/cad-libraries/issues/36#issuecomment-874661606) we still have a problem assigning the correct density to commercial components.  
Here, this value should be 0.0000037162 but Creo rounds it up to 0.000004.

[image]

So, the overall weight of The RealSense D450 **results in 32 g instead of 29,8 g**.   
For now, there is no solution.  
  
In this case, in percentage is a 7% difference with respect to the real weight but, in absolute terms, it is a really small difference, I think this issue can be closed @fiorisi.

Mick3Lozzo, 2021-07-14:
[Interposer.zip](https://github.com/icub-tech-iit/cad-libraries/files/6817296/Interposer.zip)

Found an unofficial `step` of the rigid PCB, I'll put it in `cad-libraries`.

fiorisi, 2021-07-14:
Thanks @Mick3Lozzo.

> think this issue can be closed @fiorisi.

I would not close the issue. We close it once the mass properties are correct.

fiorisi, 2021-08-05:
@Mick3Lozzo can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

Mick3Lozzo, 2021-08-06:
[image]

Done: 29,8 g.  
  
cc @fiorisi

closed 2021-08-06 by Mick3Lozzo