mesh-iit/cad-libraries #43: [Intel] - [INTEL_REALSENSE_D455] - [Intel RealSense Camera D455]

Mick3Lozzo, 2021-07-08:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [Intel_RealSense_D455_CAD_external_v2.zip](https://github.com/icub-tech-iit/cad-libraries/files/6785251/Intel_RealSense_D455_CAD_external_v2.zip), [Intel-RealSense-D400-Series-Datasheet-June-2020.pdf](https://github.com/icub-tech-iit/cad-libraries/files/6785265/Intel-RealSense-D400-Series-Datasheet-June-2020.pdf)  |
| Project              | ergoCub |
| Alias Creo           | INTEL_REALSENSE_D455   |
| IIT-Code             | XXXXX       |
| Applicant            | Michele Gesino, @Mick3Lozzo              |
| Supervisor           | Ugo Pattacini, @pattacini            |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#cad-documentation-standards).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Edited the [parameters](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

Mick3Lozzo, 2021-07-08:
*   This component still does not have an IIT-code
*   The mass should be 387 g, better to verify if we buy one.

fiorisi, 2021-07-08:
> The mass should be 387 g, better to verify if we buy one.

Which is the source of this number?

Mick3Lozzo, 2021-07-08:
My mistake, this is the weight of the complete package, I found another discussion [here](https://support.intelrealsense.com/hc/en-us/community/posts/360049877673-Weight-of-D455-Depth-Camera-).

_There is not currently an official weight for the D455 in the datasheet at the time of writing this, as you observed.  I did a very rough measuring test of a D435 and D455 with miniature kitchen scales without a tripod or USB cable attached.  I got readings of 55 grams for the D435 and 103 grams for the D455.

This weight measurement for the D455 should not at all be taken as a precise measurement but gives an impression of the weight ratio between D415 / D435 and D455._

cc @fiorisi

fiorisi, 2021-07-08:
> Now we have a computed weight of 105 g, I think it's good.

I would write to the manufacturer asking the precise wights. Meanwhile we asked to DEDA to check the mismatch between the computed and assigned masses. The issue can not be closed for the time being.

fiorisi, 2021-08-05:
@Mick3Lozzo can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

Mick3Lozzo, 2021-08-06:
[image]

Done: 103 g  
  
cc @fiorisi

closed 2021-08-06 by Mick3Lozzo