mesh-iit/cad-libraries #44: ECAD - [INTEL_REALSENSE-VISION-D4] - [XXXXX] - [Intel, RealSense, Vision Processor D4 Board]

Mick3Lozzo, 2021-07-13:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | 
[Intel_RealSense_Vision_Processor_D4_USB_Type_C_V05_PC066105.STEP_.zip](https://github.com/icub-tech-iit/cad-libraries/files/6807634/Intel_RealSense_Vision_Processor_D4_USB_Type_C_V05_PC066105.STEP_.zip), 
[Intel-RealSense-D400-Series-Datasheet-June-2020.pdf](https://github.com/icub-tech-iit/cad-libraries/files/6807637/Intel-RealSense-D400-Series-Datasheet-June-2020.pdf)  |       |
| Manufacturer info  | Intel, RealSense™ Vision Processor D4 * |       |
| Alias Creo         | INTEL_REALSENSE-VISION-D4**          |       |
| MCAD Contact       | Michele Gesino, @Mick3Lozzo    |       |
| ECAD Contact       | Marcello Savoldi, @marcellosavoldi    |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)

### Checklist:
- [ ] Added IIT-CODE.
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

Mick3Lozzo, 2021-07-14:
**Board weight**: 4 g.

Mick3Lozzo, 2021-07-14:
[image]

This is the best result we can obtain, for now, given the known problem with density. 3,7 g.

fiorisi, 2021-07-14:
I would not close the issue. We close it once the mass properties are correct.

fiorisi, 2021-08-05:
@Mick3Lozzo can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

Mick3Lozzo, 2021-08-06:
[image]

Done: 4 g  
  
@fiorisi

closed 2021-08-06 by Mick3Lozzo