mesh-iit/cad-libraries #45: [Intel] - [Intel_82635DSITR50P] - [Intel, RealSense D400 series, interposer]

Mick3Lozzo, 2021-07-15:
| Description          | Intel RealSense D400 series interposer  |
|:---------------------|:---------------------------------------|
| Datasheet component  |  [Intel_RealSense_D400_Family_Datasheet_Jan2019.pdf](https://github.com/icub-tech-iit/cad-libraries/files/6823591/Intel_RealSense_D400_Family_Datasheet_Jan2019.pdf)  |
| Project              | ergoCub   |
| Alias Creo           | Intel_82635DSITR50P    |
| IIT-Code             | XXXXX             |
| Applicant            | Michele Gesino, @Mick3Lozzo    |
| Supervisor           | Ugo Pattacini, @pattacini                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#cad-documentation-standards).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Edited the [parameters](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#parameters-for-commercial-components).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

Mick3Lozzo, 2021-07-15:
[Interposer.zip](https://github.com/icub-tech-iit/cad-libraries/files/6823672/Interposer.zip)

Absolutely **unofficial** `step` file.

fiorisi, 2021-08-05:
@Mick3Lozzo can you please update the mass properties of the part accordingly to the [new procedure](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components)?
Considering that you didn't use the new start part, simply remove the unused parameters, all the materials and follow the suggestions of [this comment](https://github.com/icub-tech-iit/cad-mechanics/issues/152#issuecomment-893291732).
Let me know if something is not clear.

Mick3Lozzo, 2021-08-10:
According to Intel, the expected weight should be 8 g.

Mick3Lozzo, 2021-08-10:
[image]

Done.

closed 2021-08-10 by Mick3Lozzo

fiorisi, 2021-08-10:
> Done.

Great @Mick3Lozzo.
Can you please add the screenshot of the entire mass properties window?

Mick3Lozzo, 2021-08-10:
[image]