mesh-iit/cad-libraries #47: Editing [general screws]-[modification req. family table 7045-7046-5933]

marcellosavoldi, 2021-08-02:
| Description   | Field                              |
|:--------------|:-----------------------------------|
| Category      | [MCAD]                        |
| Alias Creo    | [ISO7045_Z]                   |
| IIT-Code      | [several]                     |
| Applicant     | Marcello Savoldi, @marcellosavoldi            |
| Supervisor    | Luca Fiorio, @fiorisi           |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @username                          |

## Description and motivation of the changing request: <br>
[ details and reasons of modification ]

I noted in the pozidriv cross recessed pan head screws the following improvement:
- V1_6-16--_-_ISO7045_CZ = is missing in the family table, but the screw exist ( I propose to add it)
- V2-14--_-_ISO7045_CZ = It is a no-standard lenght ( can we added in the description "no standard" ?)
- V2_5-3--_-_ISO7046-1_CH = it's not exist in the Bossaard catalog and in the standard dimensions (can we think to remove it?)
- V2_5-14--_-_ISO7045_CZ = same of V2-14 .... about this component I noted that in WINGST exist as 2719 code, but the Bossard code indicated in the normalization is wrong! it refers to a set screw ( see [Bossard wbsite](https://www.bossard.com/eshop/global-en/search/?q=bn617) )
- V2_5-25--_-_ISO7045_CZ = it's not exist in the Bossaard catalog and in the standard dimensions (can we think to remove it?)
- V3-40--_-_U5933_C = they are missing in the family table, can we add them? 
PS: the V3-35--_-_U5933_C and V3-40--_-_U5933_C have a different material (S10.9) , it is indicated in the description, not in the ALIAS. for me it could be negligible.

for more detail refer to  [Bossard website](https://www.bossard.com/eshop/global-en/screws/screws-and-bolts-with-internal-drive/pozi-pan-head-machine-screws-form-z/p/81882/?category=01.100.100.10&index=2&q=%3Arelevance%3A19D20820-0A5D-4344-AF13-2431AAD21588%3APan%2BHead%3A74EC6A2E-3795-43DB-A517-12F3521785C5%3ACross%2Brecess%2BZ%2B%2528Pozidriv%2529&variants-page=1)

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

fiorisi, 2021-08-02:
@marcellosavoldi it would be nice if you can identify a reviewer for this activity. 
A member of the @icub-tech-iit/silo-mech would be fine.

marcellosavoldi, 2021-08-06:
> @marcellosavoldi sarebbe bello se riuscissi a identificare un recensore per questa attività.
> Un membro di @icub-tech-iit/silo-mech andrebbe bene.

you...

fiorisi, 2021-08-10:
> you...

Let's see if @Mick3Lozzo or @Lawproto are willing to review this issue.

Lawproto, 2021-08-11:
@fiorisi I am not sure I am getting what is going on this issue.

marcellosavoldi, 2021-08-12:
@Lawproto , try to comment the rows above and comment the single row what do you need more informations.
This could be useful also for the @icub-tech-iit/silo-mech .
thanks

Mick3Lozzo, 2021-08-12:
> *   V1\_6-16--\_-\_ISO7045\_CZ = is missing in the family table, but the screw exist (I propose to add it)

If we use it, yes.

> *   V2-14--\_-\_ISO7045\_CZ = It is a no-standard length (can we add in the description "no standard" ?)

If Bossard has it, if we can buy it, I don't feel it necessary.

> *   V2\_5-3--\_-\_ISO7046-1\_CH = it does not exist in the Bossard catalogue and in the standard dimensions (can we think to remove it?)

Is it used somewhere in the assemblies?

> *   V2\_5-14--\_-\_ISO7045\_CZ = same as V2-14 .... about this component I noted that in WINGST exist as 2719 code, but the Bossard code indicated in the normalization is wrong! it refers to a set screw ( see [Bossard wbsite](https://www.bossard.com/eshop/global-en/search/?q=bn617) )

I **noticed**.

Well, can we correct the WinGST code, and give some advice to @Fabrizio69?

> *   V2\_5-25--\_-\_ISO7045\_CZ = it does not exist in the Bossard catalogue and in the standard dimensions (can we think to remove it?)

Is it used somewhere in the assemblies?

> *   V3-40--\_-_U5933\_C = they are missing in the family table, can we add them?_  

If we plan to use them, yes.
    
>     _PS: the V3-35--_\-_U5933\_C and V3-40--_\-\_U5933\_C have a different material (S10.9), it is indicated in the description, not in the ALIAS. For me, it could be negligible

marcellosavoldi, 2021-08-12:
Thanks for the feedback

### **regarding missing screws in family table:**
Generally when I design, I try to use the screws in the family table without doing research.
From this, my proposal to insert all the existing screws in the family table.

### **regarding the screws to be removed:**
I think that the screws I have indicated do not exist, for this reason I thought they were not mounted anywhere. But if what I wrote above is true: it is possible that this has happened.
Do you have ideas to make sure nobody uses them?

marcellosavoldi, 2021-08-12:
> Inoltre, non sono ancora sicuro delle parti Creo basate su family table, né del modo in cui iCub Tech ha scelto di sfruttarle per i dispositivi di fissaggio standard.

@Lawproto ... interressante!
Potrebbe valer la pena aprire un issue in cad-mechanics ... 
se hai proposte e commenti ben accette!

fiorisi, 2021-08-17:
> Do you have ideas to make sure nobody uses them?

I am also in favor of removing the screws that we can not buy or are not standard. Nevertheless, before I would check if are used in one of the main assemblies (iCub and R1).

marcellosavoldi, 2022-09-21:
having officially switched to Torx screws, I would say to close this ISSUE and reopen it in case the problem recurs.

what do you think @fiorisi ?

closed 2022-09-21 by marcellosavoldi