mesh-iit/cad-libraries #48: [DIODES] - [DXT2011P5 & DXT2012P5] - [NPN and PNP Transistors]

maurizbo, 2021-08-03:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [link or attachment ] *               |       |
| Manufacturer info  | [Manufacturer name + serial number] * |       |
| Alias Creo         | [Alias or filename MCAD] \**          |       |
| MCAD Contact       | Marcello Savoldi, @marcellosavoldi                |       |
| ECAD Contact       | Maurizio Borgagni, @maurizbo                |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

   
[image]

###  Attachments (optional)
I attach here the datasheet related the POWERDI-5 case with all the useful mechanical info
[PowerDI5.pdf](https://github.com/icub-tech-iit/cad-libraries/files/6922746/PowerDI5.pdf)

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)
I created a single request for two component that have differences only in "electronic values" 

### Checklist:
- [ ] Added IIT-CODE.
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

marcellosavoldi, 2021-08-06:
@maurizbo , can you check the issue...
I cant find IIT-CODE and/or datasheets

marcellosavoldi, 2021-08-20:
[image]

DONE

closed 2021-08-20 by marcellosavoldi