mesh-iit/cad-libraries #49: [Misumi] - [PACK-PPSS] - [Resin Low Head Cap Screws]

Mick3Lozzo, 2021-08-10:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [P2_0219-0220_F34_IT.pdf](https://github.com/icub-tech-iit/cad-libraries/files/6962192/P2_0219-0220_F34_IT.pdf), [PACK-PEKE3-6_Creo_Parametric_6_20210810.zip](https://github.com/icub-tech-iit/cad-libraries/files/6962194/PACK-PEKE3-6_Creo_Parametric_6_20210810.zip)|             
| Project              | Generic  |
| Alias Creo           | MISUMI_ PACK-PPSS  |
| IIT-Code             | [if available]                         |
| Applicant            | Michele Gesino, @Mick3Lozzo         |
| Supervisor           | Marco Maggiali, @maggia80   |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

marcellosavoldi, 2021-08-12:
can you describe better what did you create?
could be interesting an immage and/or a screenshot of the family table for example...
thanks

Mick3Lozzo, 2021-08-12:
[image]

[image]

marcellosavoldi, 2021-08-12:
thanks @Mick3Lozzo ,
I assuming all row signed are good...
only a doubt.... off topic ....  Do we need to buy some tool to use this screws?

other curiosity.... why you didnt use a common cross recessed as following? 
for exemple:
[SPE-M5_P](https://it.misumi-ec.com/vona2/detail/221000233306/?CategorySpec=00000246929%3a%3aa%2cb)
or
[PACK-PPSA](https://it.misumi-ec.com/vona2/detail/110300243360/?CategorySpec=00000246929%3a%3aa%2cb)

the head of these I proposed you, is thinner than your...

Mick3Lozzo, 2021-08-13:
> the head of these I proposed you, is thinner than your...

Nope. 

[image]

Mine is **1.5 mm**.

> only a doubt.... off topic .... Do we need to buy some tool to use this screws?

I think a Torx screwdriver should be enough.

[image]

If we want, this one. I think is similar to a Torx one @marcellosavoldi.

marcellosavoldi, 2021-08-13:
`If we want, this one. I think is similar to a Torx one @marcellosavoldi. Maybe we already have it in the workshop?`
to be honest I dont know...

fiorisi, 2021-08-16:
@Mick3Lozzo how did you set the material properties?

Mick3Lozzo, 2021-09-06:
> [**@Mick3Lozzo**](https://github.com/Mick3Lozzo) how did you set the material properties?

For now, I've set PP, (Polypropylene). The correct material should be PPS (Polyphenilene Sulfide).  
  
[Here](https://www.makeitfrom.com/compare/Polyphenylene-Sulfide-PPS/Polypropylene-PP-Homopolymer) I found a comparison between the two materials.  
  
 

[image]

[image]

The density is really similar, if we think is better, we could think to prepare a proper \*.mtl file.  
  
Today I have tried them, and the mounting tool we have in the workshop works.

[image]

cc @icub-tech-iit/silo-mech

fiorisi, 2021-09-06:
> For now, I've set PP, (Polypropylene). The correct material should be PPS (Polyphenilene Sulfide).

@Mick3Lozzo please do not invent yet another method to set the material for commercial parts. We have a standard way documented in the guidelines, can you stick to it? 🙂

Mick3Lozzo, 2021-09-06:
@fiorisim yep, but this is a screw part of a family table, If we set the correct \*.mtl file it will work for all instances.

Mick3Lozzo, 2021-09-07:
Created and assigned `pps` material file, density set to 1.7^10-6 kg/mm3, the weight of a single screw results to be 0,4 g.

fiorisi, 2021-09-07:
> Created and assigned pps material file, density set to 1.7^10-6 kg/mm3, the weight of a single screw results to be 0,41 g.

@Mick3Lozzo I would not create a new material. Creating a material is not only about the density, there are many other parameters that must be set correctly.

Please follow the standard procedure for commercial components. If you want to change the procedure please add a point to the next @icub-tech-iit/silo-mech meeting. 
Until then please **always** follow the guidelines, do not create new way of doing things. 🙂

Mick3Lozzo, 2021-09-07:
@fiorisi done. And added a point to the next silo meeting.

fiorisi, 2021-09-07:
> @fiorisi done. And added a point to the next silo meeting.

Great. I let the maintainer @marcellosavoldi review and close the issue.

marcellosavoldi, 2021-09-13:
I am not able to understand that it was saved with the right license, so i trust it.
otherwise ok.
@Mick3Lozzo , you can close.

closed 2021-09-21 by Mick3Lozzo