mesh-iit/cad-libraries #51: ECAD - [LSM6DSL] - [12629] - [iNEMO inertial module]

marcellosavoldi, 2021-08-12:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [link or attachment ] *               |    [ST MICROELECTRONICS - iNEMO inertial module LSM6DSL.pdf](https://github.com/icub-tech-iit/cad-libraries/files/6974098/ST.MICROELECTRONICS.-.iNEMO.inertial.module.LSM6DSL.pdf)   |
| Manufacturer info  | [Manufacturer name + serial number] * |    St microelectronics - LSM6DSLTR   |
| Alias Creo         | [Alias or filename MCAD] \**          |   LSM6DSLTR    |
| MCAD Contact       | Name Surname, @username               |    @Lawproto    |
| ECAD Contact       | Name Surname, @username               |    @fgarini    |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |   @marcellosavoldi     |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

@fgarini , please added the immage of PIN1 here.... ask to @maurizbo 

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

no required

### Notes (optional)

### Checklist:
- [x] Added IIT-CODE.
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://www.ptc.com/en/support/article/CS33072).

fgarini, 2021-08-12:
[image]
Decal from Pads layout

Lawproto, 2021-08-12:
@fgarini from your image I get this:

[image]

Is it correct?

fgarini, 2021-08-12:
> @fgarini from your image I get this:
> 
>
> 
> Is it correct?

yes it is centered

Lawproto, 2021-08-12:
Just as a note, I found this [cool website](https://www.ultralibrarian.com/) for downloading CAD 3D models for electronic components, both in a mechanical format (STEP) and in a electronical format. (Orcad, Altium, Pads... you name it!)
This is what ST Microelectronics itself uses as a 3D database for its own stuff.

simeonedussoni, 2021-08-12:
here the rules for the land pattern of LGA package

[image]

snapshot is taken from this [ST Technical note](https://www.st.com/resource/en/technical_note/tn0018-surface-mounting-guidelines-for-mems-sensors--in-an-lga-package--stmicroelectronics.pdf)

marcellosavoldi, 2021-08-12:
sorry @Lawproto ,
can you check the following things?
- I noted in the parts all parameters are missing, same for the relation.
- The mass proprieties are not right.... 
- the unit are in MM Newton Second  and not mm Kg Sec
- the footprint must be above the part , it is not valid to trace it :stuck_out_tongue_winking_eye: ... you have to create it from datasheet , we use it as a check
I noted little discrepancy on the dimensions.
Remember you have to create the model using the maximum tollerances you find in the datasheet ( in this case 2.6x3.1).

[image]
[image]
[image]
[image]

marcellosavoldi, 2021-08-12:
No @Lawproto , it's very important have a maximum dimensions in the cad-libraries parts ( the same happen for the PADS libraries parts )
The best better to proceed is "import" the step file in a new parts , not open directly the step file, this can solve many things for you.
All our components are done in this way.
if you are convinced otherwise, we can discuss it.

Lawproto, 2021-08-13:
The alias here - `LSM6DSLTR` - does not match with the alias in _WinGST_:
[image]

@fgarini which is the correct one?

Lawproto, 2021-08-13:
@marcellosavoldi I am trying to recreate the footprint without referencing the 3D model, as you suggested. I am having trouble finding a `y-axis` reference in the datasheet for the three pins group, though:

[image]

What should we do in cases like this?

maurizbo, 2021-08-13:
Hi @Lawproto alias  sometimes is related to the P/N some times not, for example capacitor are 

CAP-0402-X7R-100NF-16V-10TOL

sometimes related to the P/N 

The important thing is that it is unique for each P/N in our lib and it is correctly linked

maurizbo, 2021-08-13:
@Lawproto,  anyway due to the fact that this component is not used yet on our boards I will change the alias according with the P/N to avoid future misunderstandic
 from LSM6DSL to LSM6DSL to LSM6DSL**TR**

marcellosavoldi, 2021-08-13:
> @marcellosavoldi I am trying to recreate the footprint without referencing the 3D model, as you suggested. I am having trouble finding a `y-axis` reference in the datasheet for the three pins group, though:
> 
> [image]
> 
> What should we do in cases like this?

I suppose 0.1...
[image]

marcellosavoldi, 2021-08-13:
> La [wiki](https://github.com/icub-tech-iit/mechatronics/wiki/How-to-create-a-library-component-in-Creo-Parametric) ti invita a scaricare il modello il modello ma non menziona affatto le tolleranze massime.
> 
> Se questo aspetto è così importante, il wiki dovrebbe essere modificato di conseguenza. Tuttavia, in questo modo stiamo fortemente ostacolando la comodità di avere la geometria 3D direttamente dal sito del produttore per un vantaggio troppo piccolo in cambio, secondo me.
> 
> Capisco il ragionamento alla base di questo, ma anche per i componenti meccanici - se ci pensi - la stragrande maggioranza delle dimensioni 3D sono modellate da rappresentano il valore centrale del loro intervallo di tolleranza (vedi tolleranze ISO generali). Perché utilizzare un approccio diverso in questo caso?
> 
> Quindi, sono ancora per utilizzare i 3D del fornitore così come sono, se possibile.
> 
> Qual è la tolleranza che abbiamo per i componenti posizionati in un PCB e le dimensioni del PCB, comunque?

I think is better open an issue for this...
could be interesting deepen the discussion.
In any case I think is very important for the component keep the maximum dimensions...

fiorisi, 2021-08-13:
> I think is better open an issue for this...

Yes, I agree. We can open an issue to collect all the comments/ideas on the procedure. 
Then next week we can organize a call to review the comments.

Lawproto, 2021-08-13:
I will open an issue with my comments ASAP.
@fiorisi @marcellosavoldi

fiorisi, 2021-08-13:
@Lawproto can you check also the missing boxes before closing the issue?

[image]

Lawproto, 2021-08-13:
@fiorisi OK.

1. `Added details about PLUG connector` seems not proper for this component - it is not a connector. Should I check it, anyway?
1. I am not sure about `Converted the files with the EDU-COM license`'s meaning. Can you help me?

marcellosavoldi, 2021-08-16:
> @fiorisi OK.
> 
> 1. `Added details about PLUG connector` seems not proper for this component - it is not a connector. Should I check it, anyway?

No, you can miss it

> 2. I am not sure about `Converted the files with the EDU-COM license`'s meaning. Can you help me?

[image]
[image]
[image]

Now save the part and remember to unsign the licences EDU_Com Convert after saving.

fiorisi, 2021-08-16:
Thanks @marcellosavoldi. I've copied the images in a dedicated [guide](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

I will also update the issue templates to point the guide.

closed 2021-08-16 by Lawproto