mesh-iit/cad-libraries #54: Editing [alias]-[short description of the change]

Elio75, 2021-08-31:
| Description   | Field                              |
|:--------------|:-----------------------------------|
| Category      | [MCAD]                        |
| Alias Creo    | [CSD-14-100-2A-R / CSD-17-100-2A-R]                   |
| IIT-Code      | 15479 / 15483]                     |
| Applicant     | Elio Massa, @Elio75             |
| Supervisor    | Luca Fiorio, @fiorisi             |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @username                          |

## Description and motivation of the changing request: <br>
[ The HD 2A-GR version for sizes CDS-14 and CDS-17 does not exist. You need to change the alias to version 2A-R ]

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Elio75, 2021-08-31:
Each HD consists of 3 parts
CSD-14-100-2A-R_A / CSD-17-100-2A-R_A
CSD-14-100-2A-R_B / CSD-17-100-2A-R_B
CSD-14-100-2A-R_B / CSD-17-100-2A-R_C

To create a new alias I made a save copy of the part by editing the new version. On the previous version I created a sketch with the word OBSOLETE in order to replace the old component with the new one using the option **Replace select component** with CREO.

For example
[image]

Elio75, 2021-08-31:
### List of assemblies where the HD components to be replaced are located

**CSD-14-100-2A-GR**   (A , B , C)                 ICUB_FBBBBEBA

| Alias | Component | Quantity | 
|:---:|:---:|:---|
| **RC_IIT_018_G_073** | A | 1 |  
|RC_IIT_018_G_072  | B | 1 |  
| RC_TLR_001_G_003 | C | 1 |  
|**RC_TLR_003_A_003_RIGHTSHOULDER** |  |  |  
| RC_IIT_003_G_022 | A/C | 2 | 
| RC_IIT_003_G_021 | B | 2 |         
| **RC_TLR_005_A_001_RIGHT_UPPERARM/RC_TLR_006_A_001_LEFT_UPPERARM** |  |  | 
| RC_TLR_006_G_003 | A/C  | 2 |  
| RC_IIT_003_G_038 | B  | 2 |  
| **RC_TLR_002_A_003_LEFTSHOULDER** |  |  |  
| RC_IIT_002_G_022 | A/C | 2 |  
| RC_IIT_003_G_045 | B | 2 |  
|  |  |  |  

**CSD-17-100-2A-GR**  (A , B , C)                ICUB_FBBBBEBA

| Alias | Component | Quantity | 
|:---:|:---:|:---|
| RC_IIT_018_G_060 | A | 2|  
| RC_IIT_018_G_004 | C | 10 |  
| RC_IIT_018_G_055 | B | 10 |  
|RC_IIT_018_G_063  | A | 2 |  
| RC_IIT_018_G_059 | B | 2 |  
| RC_IIT_018_G_061 | A | 2 |  
| RC_IIT_018_G_062 | A | 1 |  
| RC_IIT_018_G_101 | B | 2 |  
| RC_IIT_018_G_065  | A | 1  |  
| RC_IIT_002_G_024 | A/C | 2 |  
| RC_IIT_003_G_046 | B | 2 |  
| RC_IIT_018_G_067 | A | 1 |  
| RC_TLR_001_G_023 | C | 6 |  
| RC_IIT_018_A_016_LOWERTORSO | A | 4 |  
| RC_IIT_018_G_069 | A | 1 |

Elio75, 2021-08-31:
Cod WGST

CSD-14-100
[image]

CSD-17-100
[image]

Elio75, 2021-09-01:
### Commit made

Aggiunto nuovo componente HD [CSD-14-100-2A-R](https://github.com/icub-tech-iit/cad-libraries/commit/e92c694a061ce4ae4f443655075142afc5befb8f)
Aggiunto nuovo componente HD [CSD-17-100-2A-R](https://github.com/icub-tech-iit/cad-libraries/commit/92cebb79386c3975c7d7568eedd56364045bdba4)
Save [Edu_com](https://github.com/icub-tech-iit/cad-libraries/commit/9298ededce8d244f84ea04365962b388b5245659)
Replacement CSD-14-100-2A-GR (A,B,C) in [CSD-14-100-2A-R (A,B,C)](https://github.com/icub-tech-iit/cad-mechanics/commit/0209f4266c4294b7034057d6fd9a9437c30f2252)
Replacement CSD-17-100-2A-GR (A,B,C) in [CSD-17-100-2A-R (A,B,C)](https://github.com/icub-tech-iit/cad-mechanics/commit/a49601e2cfe360d288ae9b91399c631a20dac79a)

fiorisi, 2021-09-10:
@Elio75 can you thoroughly review the task with a member of the @icub-tech-iit/silo-mech ?

Elio75, 2021-09-10:
ok I ask @Mick3Lozzo if he can

marcellosavoldi, 2021-09-13:
can I propose to add a second row in the written on the model with "replaced by..."..
what do you think?

Elio75, 2021-09-13:
Yes, why not, even if you search for it on Wingst you will be told by who it is replaced

@marcellosavoldi

Mick3Lozzo, 2021-09-13:
The writings in the model are useless, they are not seen by WinGST and they don't automatically upgrade the BOMs. They are visible only at a CAD level, but they are meant to be used by someone who does NOT use the CAD.

My opinion.  
  
cc @Fabrizio69

marcellosavoldi, 2021-09-13:
I think that the writing on the model very convenient to understand that an obsolete object exists in that assembly.
the additional row with "replaced by ..." was just an idea to find the replacement first, but I agree that the information can be found on Wingst.

fiorisi, 2021-09-23:
> I think that the writing on the model very convenient to understand that an obsolete object exists in that assembly.
the additional row with "replaced by ..." was just an idea to find the replacement first, but I agree that the information can be found on Wingst.

I would keep the `OBSOLETE` sketch, but not the "replaced by".

fiorisi, 2021-12-15:
@Elio75 if you have to make some components obsolete, remember to open a dedicated issue using the template issue:
[image]

Elio75, 2022-06-22:
Yes

closed 2022-06-22 by Elio75