mesh-iit/cad-libraries #55: Editing CUSCI_DGBB1.PRT-Adding new variants W_61705,W_61706

divyashah, 2021-09-02:
| Description   | Field                              |
|:--------------|:-----------------------------------|
| Category      | MCAD                       |
| Alias Creo    | CUSCI_DGBB1.PRT                  |
| IIT-Code      | [if available]                     |
| Applicant     | Divya Shah, @divyashah             |
| Supervisor    | Luca Fiorio, @fiorisi            |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @icub-tech-iit/silo-mech                         |

## Description and motivation of the changing request: <br>
For the conceptual design of the new ergoCub joint, it is desired to have narrow width bearings. For this reason I am using the following bearings from SKF: 
- [SKF_W-61705](https://www.skf.com/group/products/rolling-bearings/ball-bearings/deep-groove-ball-bearings/productid-W%2061705) : **25x32x4**
- [SKF_W-61706](https://www.skf.com/group/products/rolling-bearings/ball-bearings/deep-groove-ball-bearings/productid-W%2061706) : **30x37x4**

These variants are missing from the family table in `CUSCI_DGBB1.PRT`. I would like to add these variants within the family table. 

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

fiorisi, 2021-09-10:
Ok for me. Review the alias and the family table instances with @marcellosavoldi.

divyashah, 2021-09-16:
> - [SKF_W-61705](https://www.skf.com/group/products/rolling-bearings/ball-bearings/deep-groove-ball-bearings/productid-W%2061705) : **25x32x4**
> - [SKF_W-61706](https://www.skf.com/group/products/rolling-bearings/ball-bearings/deep-groove-ball-bearings/productid-W%2061706) : **30x37x4**

In addition to these two instances mentioned above in the description, I decided to add also a third instance that could be useful later in the designs: 

- [SKF_W-61707](https://www.skf.com/group/products/rolling-bearings/ball-bearings/deep-groove-ball-bearings/productid-W%2061707) : **35x44x5**

Based on the specifications mentioned in the links, I modified the family table of the part as shown below: 

[image]

divyashah, 2021-09-16:
I noticed that we family table has a column to add mass (`PESO`) given in the datasheet. However the mass of the component computed from the Creo `Mass Properties` differs from this family table value. 

As an example, for Bearing **W 61705** the datasheet mass is **~6e-3 kg** whereas the Creo mass turns out to be **~6.77e-3 kg**.  The difference, in this case, is negligible. 
| [image] |
|:----:|
| Mass Properties for **W61705** for default material |

Should we can consider assigning the generic part a particular material/density maybe **stainless steel**?? 

But with stainless steel (`X5-CR-NI-MO1712-AS136`) I get a greater mass value (**~8.3e-3 kg**). 

| [image] |
|:----:|
| Mass Properties for **W61705** for `X5-CR-NI-MO1712-AS136` |

@icub-tech-iit/silo-mech Do you usually verify the mass properties for the bearings of our family table ??

marcellosavoldi, 2021-09-17:
`Should we can consider assigning the generic part a particular material/density maybe stainless steel??`

For me this is a good solution!
most bearings we are using,  are in stainless steel.
The balls of the bearing are represented as a circular tube... I think  can negligible the different of weight..
... personal opinion ....

fiorisi, 2021-09-23:
@divyashah how is used the parameter `PESO`? Is there a relation?

marcellosavoldi, 2021-09-23:
... I would like, one day, to remove all Italian words and therefore most of the relations ...

divyashah, 2021-09-24:
> @divyashah how is used the parameter `PESO`? Is there a relation?

No @fiorisi, I do not see any particular relation involving the parameter `PESO`. 

[image]

divyashah, 2021-09-27:
With the previous commit, I added the instances of `W61705`, `W61706`, `W61707` within the family table of `CUSCI_DGBB1.PRT`. 
I kept the mass properties like default for now. 

We can then decide if we want to assign a particular material to  family table or not during the next @icub-tech-iit/silo-mech  meeting.

fiorisi, 2021-09-27:
> We can then decide if we want to assign a particular material to family table or not during the next @icub-tech-iit/silo-mech meeting.

Yes I would discuss it during the silo meeting.

marcellosavoldi, 2021-09-30:
For me we can added the new variants to cad-libraries.
We need of the parts to continue the joint study...
After we can modify the parameters as we will go to decide.
what do you think @icub-tech-iit/silo-mech ?

divyashah, 2021-09-30:
> For me we can added the new variants to cad-libraries. We need of the parts to continue the joint study... After we can modify the parameters as we will go to decide. what do you think @icub-tech-iit/silo-mech ?

Yes @marcellosavoldi. I already added the new variants to `CUSCI_DGBB1.PRT` which is the bearing type without any shields. About the mass properties/material we can modify later as we decide.

fiorisi, 2021-09-30:
> For me we can added the new variants to cad-libraries.
We need of the parts to continue the joint study...
After we can modify the parameters as we will go to decide.
what do you think @icub-tech-iit/silo-mech ?

I agree. If the mass is correct, I think that we cna close the issue.
The material for family tables will be discussed during the next silo meeting.

divyashah, 2021-09-30:
> I agree. If the mass is correct, I think that we can close the issue. The material for family tables will be discussed during the next silo meeting.

The mass I obtain from the default properties are slightly higher than the datasheet mass. 

| Bearing | Datasheet mass | CREO mass | 
| :----: | :----: | :----: | 
| W61705 | 6e-3 kg | 6.77e-3 kg | 
| W61706 | 7e-3 kg | 8.06e-3 kg | 

I think this comes from the fact that the CREO feature for the family table has a tube (in the left part) to represent the spherical balls (as on the right part). 
[image]

cc @fiorisi

marcellosavoldi, 2021-10-04:
@divyashah , we can use the body to define 2 different material ( for examble Ergal for the "tube" ), but I think we can discuss this in the next future meeting... Push for now.