mesh-iit/cad-libraries #56: SEEGER - SW-25 - Snap ring type SW for shafts

divyashah, 2021-09-02:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [SW rings for shaft](https://www.seeger-orbis.com/products/snap-rings/circlips/snap-rings/circlips-to-din-9927-/-9928)                  |
| Project              | Generic    |
| Alias Creo           | SEEGER_SW-25                       |
| IIT-Code             | [if available]                         |
| Applicant            | Divya Shah, @divyashah                |
| Supervisor           | Luca Fiorio, @fiorisi                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |

### Attachment:
[Seeger_SW_Datasheet.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7099781/datenblatt_seeger-ringe_Sprengringe_Wellen.pdf)

### Notes:
For the new ergoCub joint, I am using the **SEEGER_SW-25** snap ring for shaft. So I would like to add it to the library. 
However, we can consider creating a **family table** of this category of snap rings, since this is missing. 

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [x] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [x] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

fiorisi, 2021-09-10:
@divyashah concerning the alias of these parts I would not use SEEGER_xxx.
For this kind of fasteners (like screws and bolts) we use the technical specification standard. We have to check the one that defines this components and if available in more than one decide which use.
I've opened a [dedicated page](https://github.com/icub-tech-iit/cad-mechanics/wiki/Technical-specification-standards).

marcellosavoldi, 2021-09-13:
in the past we looking for new componets in this [folder](https://istitutoitalianotecnologia.sharepoint.com/:f:/r/sites/icub-facility/Documenti%20condivisi/Silo%20Mechanics/sandbox/proe___librerie_compass?csf=1&web=1&e=2b1AMH) ...
Don't ask me where these documents come from because I don't remember

divyashah, 2021-09-14:
> @divyashah concerning the alias of these parts I would not use SEEGER_xxx.
> For this kind of fasteners (like screws and bolts) we use the technical specification standard. We have to check the one [...]

I am using the `SW` type of the Seeger Ring as mentioned in the description. On the Seeger Rings website, the standard mentioned for this type of rings is **DIN 9927**. 

However, this type of elastic ring is not mentioned among the brochures that are linked in the technical standard specification page on the wiki referenced above. 

> in the past we looking for new componets in this [folder](https://istitutoitalianotecnologia.sharepoint.com/:f:/r/sites/icub-facility/Documenti%20condivisi/Silo%20Mechanics/sandbox/proe___librerie_compass?csf=1&web=1&e=2b1AMH) ...

Also, from the files present in this sharepoint folder, the ring similar to one I am using is named `NOR_INA_WR.PRT` which doesn't have any particular standard code (ISO/UNI/DIN). 

We need to ensure if this is actually a standard ring or not. We can then define the alias accordingly. 

cc @marcellosavoldi @fiorisi

fiorisi, 2021-12-06:
@divyashah can we close the issue following the approved [technical standards](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#technical-standards)?

divyashah, 2021-12-07:
> @divyashah can we close the issue following the approved [technical standards](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#technical-standards)?

@fiorisi I still did not commit the file for the this. I have created the family table, but I wanted to check if I can somehow assign the calculated density, for every instance, as a relation.

I will try it and then commit the files accordingly, before closing this.

marcellosavoldi, 2021-12-09:
since we're talking about an object made of a material ... can we just assign a 308 stainless steel and let Creo do it all?

@fiorisi , @divyashah

fiorisi, 2021-12-10:
> since we're talking about an object made of a material ... can we just assign a 308 stainless steel and let Creo do it all?

If the datasheet of the component specifies that is made of that material, then yes.
Otherwise I would wait to have the new materials library and discuss weather introducing some generic materials to use for these cases.

MrAndrea, 2022-03-07:
It will be useful to discuss this task at the next Silo Meeting at the of the month.

divyashah, 2022-06-15:
I added the `DIN_9927.PRT` family table for the SEEGER ring type **SW** for shafts. 
[image]

For now, the family table consists of instances from nominal diameter 14 mm to 47 mm. 
[image]

I also assigned calculated density for each instance following the solution proposed in https://github.com/icub-tech-iit/cad-mechanics/issues/214#issuecomment-1155507804.

closed 2022-06-29 by fiorisi