mesh-iit/cad-libraries #57: Misumi - CPLCN32-9.525-RDC14 - Flexible shaft coupling

divyashah, 2021-09-02:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [Misumi Link](https://it.misumi-ec.com/vona2/detail/110300126210/?ProductCode=CPLCN32-9.525-RDC14)                  |
| Project              | Generic     |
| Alias Creo           | MISUMI_CPLCN32-9_525-RDC14-B.prt                       |
| IIT-Code             |                         |
| Applicant            | Divya Shah, @divyashah                |
| Supervisor           | Luca Fiorio, @fiorisi               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |

### Attachment:
- CAD source file downloaded from Misumi:  [CPLCN32-9.525-RDC14_Creo_Parametric_6_20210721.zip](https://github.com/icub-tech-iit/cad-libraries/files/7100181/CPLCN32-9.525-RDC14_Creo_Parametric_6_20210721.zip)

- CAD render of the component 
[image]

### Notes:
This flexible shaft coupling is to be used to attach the motor group prototype of the ergoCub joint to the brake test bench for testing the motors. Thus I would like to add it to our library. 

### Checklist:
Complete the checklist in order to proceed with the request.

- [x] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [x] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [x] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

fiorisi, 2021-09-03:
cc @icub-tech-iit/silo-mech

divyashah, 2021-09-03:
> @divyashah can you double check the alias? I can not find the product `CPLCN32-9_525-RDC14-B`:

The MISUMI product code is `CPLCN32-9.525-RDC14` as in the title of this issue. Product is also linked in the description. 

I then named the Creo part as `MISUMI_CPLCN32-9_525-RDC14-B.prt`, based on the part name of the source CAD file. 
Let me know if this part name is ok, or I should update it ?? 
cc @fiorisi

fiorisi, 2021-09-03:
Ok I see. 
Probably for codes that contain a `.` we can decide to replace it with `_`.

In this case I would add few lines in the [guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).

divyashah, 2021-09-03:
> Probably for codes that contain a `.` we can decide to replace it with `_`.
> 
> In this case I would add few lines in the [guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).

Yes, I will update the guidelines accordingly.