mesh-iit/cad-libraries #59: Editing [NVIDIA_JETSON_AGX_XAVIER_HS_CB]-[updating mass properties]

Mick3Lozzo, 2021-09-10:
| Description   | Jetson AGX Xavier with heatsink and carrier board      |
|:--------------|:-----------------------------------|
| Category      | [ECAD-MCAD]                        |
| Alias Creo    | NVIDIA_JETSON_AGX_XAVIER_HS_CB    |
| IIT-Code      | 14615, 14616, 14617  |
| Applicant     | Michele Gesino, @Mick3Lozzo    |
| Supervisor    | Ugo Pattacini, @pattacini     |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @icub-tech-iit/silo-mech           |

## Description and motivation of the changing request: <br>
Updating mass properties, following the defined procedure.

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Mick3Lozzo, 2021-09-10:
Ask @emilianob80 to weigh the only assembled group we have in IIT, at the moment mounted on an R1 robot.

[image]

Mick3Lozzo, 2021-09-10:
Opened [ticket](https://github.com/icub-tech-iit/tickets/issues/1634).

fiorisi, 2021-09-21:
> Opened ticket.

@Mick3Lozzo please document here the picture of the weight scale with the component.

emilianob80, 2021-09-21:
> > Opened ticket.
> 
> @Mick3Lozzo please document here the picture of the weight scale with the component.

@Mick3Lozzo @fiorisi  sorry it is my fault that due to lack of time I was unable to satisfy the open ticket. Tomorrow morning I will carry out the weight if the robot is available

Fabrizio69, 2021-09-21:
Sorry @Mick3Lozzo , if you need of the PROTO-Team support, mention me too. Thanks

cc: @emilianob80 @fiorisi

pattacini, 2021-09-21:
Yes, it's important to keep the SM in the loop.

emilianob80, 2021-09-22:
@Mick3Lozzo the robot is connected and completely switched on (PC and motors) so it is not possible for me to remove the cards and carry out the weighing

cc: @Fabrizio69 @pattacini @fiorisi

pattacini, 2021-09-22:
I went down to check the system and I just found the SN003 on and plugged into the power supply. There's no console around.
@randaz81, @datspace @lnobile, can we power it off using the switches as it is?

As it's written here, we'd need to temporarily unmount/mount the NVIDIA board to measure the actual weight.

DatSpace, 2021-09-22:
Yes certainly. Everything is saved and the robot is not running anything really. It was simply recommended to not be killed to increase the lifespan of the electronics.

lnobile, 2021-09-22:
Yeah no problem also for me! Will the robot be available tomorrow? (With the NVIDIA board mounted)

pattacini, 2021-09-22:
Thanks heaps guys 👍🏻 
So, @emilianob80, we can proceed by switching off the robot as it is.
Please, just coordinate w/ @lnobile to address his request below and agree on when it's the right moment.

>Will the robot be available tomorrow? (With the NVIDIA board mounted)

emilianob80, 2021-09-22:
@lnobile @pattacini the robot is available again. Thanks

cc: @Fabrizio69 @DatSpace @randaz81

pattacini, 2021-09-22:
Super 🚀

Mick3Lozzo, 2021-09-23:
[image]

[image]

Updated density and corrected mass.

Mick3Lozzo, 2021-09-23:
Converted with EDU-COM license. I'll close the issue.

closed 2021-09-23 by Mick3Lozzo