mesh-iit/cad-libraries #61: [RLS] - [MRA7D049AB025E00] - [Magnetic wheel for AksIM encoder]

marcellosavoldi, 2021-09-21:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [website](https://www.rls.si/eng/aksim-rotary-absolute-encoder-module )            |
| Project              | ergoCub     |
| Alias Creo           | rls_MRA7D049AB025E00                        |
| IIT-Code             | 11259                       |
| Applicant            | Marcello Savoldi, @marcellosavoldi                |
| Supervisor           | Luca Fiorio, @fiorisi               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

marcellosavoldi, 2021-09-21:
dimensions:
[image]
[image]

divyashah, 2021-09-21:
For the ergoCub joint we previously discussed about the encoders to use in icub-tech-iit/ergocub-design-joint#7.  

As mentioned by @MrAndrea, it was decided to use AksIM-2 with the following two magnetic rings:  
> `MRA049BC025DSE00` with Countersunk fasteners
> `MRA049BG034DSN00` with Glue

The corresponding purchase request is icub-tech-iit/purchases#1307. 

> I'm going to create a family table where I'll add all variant we'll need.

@marcellosavoldi  if you are creating a family table of these magnetic rings, then I think it would be useful to add also these two configurations, `MRA049BC025DSE00` and `MRA049BG034DSN00`.

marcellosavoldi, 2021-09-22:
I proceed to add in the family table the following encoders:
[image]
[image]
[image]

fiorisi, 2022-06-22:
@marcellosavoldi once the [checklist](https://github.com/icub-tech-iit/cad-libraries/issues/61#issue-1002654124) has been completed can we close this issue?

marcellosavoldi, 2022-06-27:
we decided to not use a family table creating the parts.
we can close

closed 2022-06-27 by marcellosavoldi

fiorisi, 2022-06-27:
> we decided to not use a family table creating the parts. we can close

Thanks, can you just complete the [checklist](https://github.com/icub-tech-iit/cad-libraries/issues/61#issue-1002654124)?