mesh-iit/cad-libraries #63: Editing [HFUC-5-50-2A-R-SP_A]-[updating mass]

Mick3Lozzo, 2021-09-29:
| Description   | Harmonic drive - Gearbox, size 5, gear ratio 50  |
|:--------------|:-----------------------------------|
| Category      | [ECAD-MCAD]                        |
| Alias Creo    | HFUC-5-50-2A-R-SP_A, HFUC-5-50-2A-R-SP_B, HFUC-5-50-2A-R-SP_C   |
| IIT-Code      | 2708.1       |
| Applicant     | Michele Gesino, @Mick3Lozzo         |
| Supervisor    | Ugo Pattacini, @pattacini    |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @icub-tech-iit/silo-mech           |

## Description and motivation of the changing request: <be>

I have found a mistake in the  HFUC-5-50-2A-R-SP harmonic drive's mass of 36 g but, according to the datasheet, it should be 9 g.

[image]

[image]

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Mick3Lozzo, 2021-09-29:
[image]

Ok, the problem was the wave generator, its density had ten times the expected value.

Mick3Lozzo, 2021-09-29:
Done.

marcellosavoldi, 2021-10-04:
If the part is ok, @Mick3Lozzo you can close and push the prt!

closed 2021-10-04 by Mick3Lozzo