mesh-iit/cad-libraries #64: [Slamtec] - [RPLIDAR_USB_ADAPTER] - [RPLidar USB Adapter Board for A2/A3]

Mick3Lozzo, 2021-09-29:
| Description          | Slamtec RPLidar USB Adapter Board for A2/A3  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [LM310_SLAMTEC_rplidarkit_usermanual_A3M1_v1.0_en.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7251913/LM310_SLAMTEC_rplidarkit_usermanual_A3M1_v1.0_en.pdf)   |
| Project              | R1, ergoCub    |
| Alias Creo           | SLAMTEC_RPLIDAR_USB-ADAPTER-A2-A3     |
| IIT-Code             | ---                   |
| Applicant            | Michele Gesino, @Mick3Lozzo                 |
| Supervisor           | Marco Maggiali, @maggia80      |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Mick3Lozzo, 2021-09-29:
[image]

Mick3Lozzo, 2021-09-29:
[image]

fiorisi, 2021-09-29:
@Mick3Lozzo the manufacturer doesn't have a code for this product? 
Or `RPLIDAR-ADAPTER-A2-A3` is the manufacturer code?

If we don't know, could be worth to send an email to retrieve it.

Mick3Lozzo, 2021-09-29:
@fiorisi I haven't found anything more specific on the website, we can ask.

Mick3Lozzo, 2021-09-30:
Found

[image]

Mick3Lozzo, 2021-09-30:
[image]

.... well, technically we should rename ALL RPLidar items we are using in the assemblies.

fiorisi, 2021-09-30:
Great!

Why do you keep `RPLIDAR` ? Is not part off the product code. If we stick to the [guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard) we shouldn't include it, right?

Mick3Lozzo, 2021-10-04:
> Why do you keep `RPLIDAR`? Is not part of the product code. If we stick to the [guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard) we shouldn't include it, right?

With other members of this product family, we didn't use the reseller code at all but there is a reason: on the [official product page](https://www.slamtec.com/en/Lidar/A3) of Slamtec there doesn't seem to be mention of the USB adapter as a single product - or at least I haven't found it. 
In every other component of the RPLIDAR family, we didn't use this reseller product code because we found a corresponding code on the Slamtec website. 

[image]

It is mentioned in the attached pdf manual, but it doesn't seem to have a specific code, it is only called "RPLIDAR Adapter". Moreover, according to the manufacturer website, it **seems** this item can be purchased only as a part of a development kit. The reseller Roboshop, instead, allows you to buy it separately.

So, the choice in my opinion could between these two codes:

- `SLAMTEC_RPLIDAR-ADAPTER-A2-A3` (the previous one)
- `SLAMTEC_RB-RPK-10`

@fiorisi

marcellosavoldi, 2021-10-04:
I would propose to use the name of the company that sells the item.
As I seem to have understood that Samtec does not sell this item individually ...

closed 2021-12-06 by fiorisi