mesh-iit/cad-libraries #65: Editing [SLAMTEC_RPLIDAR-A2M4-R1]-[correcting mass]

Mick3Lozzo, 2021-09-30:
| Description   | Field                              |
|:--------------|:-----------------------------------|
| Category      | [ECAD-MCAD]                        |
| Alias Creo    | SLAMTEC_RPLIDAR-A2M4-R1                |
| IIT-Code      | 11404                  |
| Applicant     | Michele Gesino, @Mick3Lozzo             |
| Supervisor    | Marco Maggiali, @maggia8    |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @icub-tech-iit/silo-mech           |

## Description and motivation of the changing request: <br>
Correcting the mass value.

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Mick3Lozzo, 2021-09-30:
[image]

Done.

Mick3Lozzo, 2021-09-30:
Note: 199 g is a value obtained by weighting a `15507 -SLAMTEC_RPLIDAR-A3M1` model, I expect that there is no significant difference, but we should verify.

fiorisi, 2021-12-06:
Gentle reminder for @Mick3Lozzo.

Mick3Lozzo, 2021-12-09:
[image]

[image]

@fiorisi on the website, RPLIDAR is used only with some components, I would consider it as part of the component's name.

fiorisi, 2021-12-10:
In the [datasheet](https://bucket-download.slamtec.com/bd39cf1da6f2c64fdb178440106561dc9fc0aa6d/LD310_SLAMTEC_rplidar_datasheet_A3M1_v1.9_en.pdf) of this component there is written `A3M1`. I think that `RPLIDAR` is the series. If you already used the part in many assemblies then we we can avoid the renaming.

[image]

Mick3Lozzo, 2021-12-13:
@fiiorisi  already used It only in the ergoCub head, but there are other RPLIDAR models in the various R1 versions.

[image]

.... and they have `RPLIDAR` in the filename.

fiorisi, 2021-12-14:
> @fiiorisi  already used It only in the ergoCub head, but there are other RPLIDAR models in the various R1 versions.

This is work done before the new guidelines and we do not modify it.

Mick3Lozzo, 2021-12-16:
@fiorisi errata corrige: this is an **OLD** component too, and is present in R1 assemblies, added before the new guidelines.

The new RPLIDAR used in ergoCub head is the `SLAMTEC_RPLIDAR_A3M1`, which I could rename in `SLAMTEC_A3M1`.

closed 2021-12-21 by fiorisi