mesh-iit/cad-libraries #66: [Slamtec] - [RPLIDAR A3M1 RB-Rpk-07] - [Laser Scanner 25 m]

Mick3Lozzo, 2021-09-30:
| Description          | RPLidar A3M1 Laser Scanner     |
|:---------------------|:---------------------------------------|
| Datasheet component  | 
[rplidar-a2m4-r1.2d.en-us.PDF](https://github.com/icub-tech-iit/cad-libraries/files/7260265/rplidar-a2m4-r1.2d.en-us.PDF) |
| Project              | ergoCub, R1   |
| Alias Creo           | SLAMTEC_RPLIDAR-A3M1     |
| IIT-Code             | 15507                      |
| Applicant            | Michele Gesino, @Mick3Lozzo        |
| Supervisor           | Marco Maggiali, @maggia80     |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

It has the SAME geometry and datasheet than A2 version.

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

fiorisi, 2021-09-30:
@Mick3Lozzo can you check the alias? As in https://github.com/icub-tech-iit/cad-libraries/issues/64#issuecomment-931407338, please check if it is compliant with the guidelines. In particular, is `RPLIDAR` part of the producer code?

Mick3Lozzo, 2021-10-04:
@fiorisi it is a reseller code, my mistake to include it in the issue title.

closed 2021-12-06 by fiorisi