mesh-iit/cad-libraries #69: NVIDIA - jetson agx xavier- [short description]

salvi-mattia, 2021-10-20:
| Description          |NVIDIA - jetson agx xavier                          |
|:---------------------|:---------------------------------------|
| Datasheet component  | https://developer.nvidia.com/embedded/jetson-agx-xavier-developer-kit            |
| Project              | ergocub et al    |
| Alias Creo           | NVIDIA_JETSON_AGX_XAVIER                        |
| IIT-Code             | 14617                     |
| Applicant            | @salvi-mattia            |
| Supervisor           | @pattacini               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

[image]

### Notes:
I didn't find weight in any datasheet, just on [this](https://forums.developer.nvidia.com/t/weigth-of-jetson-agx-xavier-module-only/71381) forum.

PLEASE DO NOT CLOSE THIS ISSUE BEFORE CHECKING THE WEIGHT ON A SCALE

### Checklist:
Complete the checklist in order to proceed with the request.

- [X] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [X] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [X] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [x] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [x] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [x] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

fiorisi, 2021-12-06:
@salvi-mattia great.
Please just align the code accordingly with [the guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard). Then we can close.

fiorisi, 2021-12-15:
> @salvi-mattia great.
Please just align the code accordingly with the guidelines. Then we can close.

Gentle reminder for @salvi-mattia.

fiorisi, 2022-06-22:
@salvi-mattia can we close this issue?

salvi-mattia, 2022-06-22:
no,
ther is still the pending decision if you want to change the name even if it is used somewhere
[is it really used somewhere?]

closed 2022-06-22 by salvi-mattia