mesh-iit/cad-libraries #74: [BOSSARD] - [MANY] - [Torx screw]

fiorisi, 2021-11-02:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | [where will the component be used]     |
| Alias Creo           | [maker + code]                         |
| IIT-Code             | [if available]                         |
| Applicant            | Marcello Savoldi, @marcellosavoldi                |
| Supervisor           | Name Surname, @username                |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

marcellosavoldi, 2021-11-03:
Below the list of the types of screws with a hexalobe footprint that we are going to use:
### ISO14583 
[image]
- Hexalobular (6 Lobe) socket pan head machine screws 
- fully threaded , 
- stainless steel, 
- type A4
### ISO14580
[image]
- Hexalobular (6 Lobe) socket head cap screws with low head, 
- fully threaded , 
- stainless steel, 
- type A2
- Class 70
### ISO14579
[image]
- Hexalobular (6 Lobe) socket head cap screws 
- fully threaded , 
- stainless steel, 
- type A2
- Class 70
## ISO14581
[image]

- Hexalobular (6 lobe) socket countersunk flat head screws 
- fully threaded
- stainless steel, 
- type A2
- Class 70

_________________________________________________________________________________________________
_________________________________________________________________________________________________
_________________________________________________________________________________________________

... specifically we will have:

## cylindrical head:
M1.6 = ISO14583 A4
M2 = ISO14583 A4
M2.5 = ISO14583 A4
M3 = ISO14583 A4 and ISO14580 A2-70
M4 = ISO14580 A2
M5 = ISO14580 A2
M6 = ISO14580 A2
M8 = ISO14579 A2
M10 = ISO14579 A2
## countersink
M1.6 = missing
M2 ... M10 = ISO14581 A2

marcellosavoldi, 2021-11-10:
Following the table with all screws we will go to use....
I have to check the Wurth  availability..
[viti-torx-da-ordinare_rev_2.xlsx](https://github.com/icub-tech-iit/cad-libraries/files/7513616/viti-torx-da-ordinare_rev_2.xlsx)
.

fiorisi, 2021-11-11:
Great job @marcellosavoldi!
Once we are ready I think that this table should be placed in the wiki, let me know if you need help.
Plobably we can add a point in the [list](https://github.com/icub-tech-iit/cad-libraries/wiki#lists) section.

marcellosavoldi, 2021-11-11:
> Great job @marcellosavoldi! Once we are ready I think that this table should be placed in the wiki, let me know if you need help. Probably we can add a point in the [list](https://github.com/icub-tech-iit/cad-libraries/wiki#lists) section.

Definitely a good idea @fiorisi .
I'd like to add a section where we indicate the "favorite library items" for our projects, such as screws, nuts, inserts, glues etc ...

Most likely I would like help from the @icub-tech-iit/silo-mech  to normalize everything on Wingst .... it's a very long job!

I also heard from @Miche19 , and he's already setting up a warehouse space for the Torx screws (cc: @Fabrizio69 )

marcellosavoldi, 2021-11-16:
I am awaiting an answer from Wurth to understand the exact availability of all screws in the catalog.
I attach the updated file with the list of screws used.
In the "full-TORX-list" TAB you will find the list of all the screws that I would like to normalize.
[viti-torx-da-ordinare_rev_2.xlsx](https://github.com/icub-tech-iit/cad-libraries/files/7548155/viti-torx-da-ordinare_rev_2.xlsx)
In the meantime I have prepared the family table models of the screws:
[torx_iso14581-14579.prt](libraries/fasteners/torx_iso14580-14579.prt.1)
[torx_iso14581.prt](libraries/fasteners/torx_iso14581.prt.1)
[torx_iso14583.prt](libraries/fasteners/torx_iso14583.prt.1)

[image]

fiorisi, 2021-11-17:
> In the meantime I have prepared the family table models of the screws:

Can we ask to DEDA? They might already have it.

marcellosavoldi, 2021-11-17:
> > In the meantime I have prepared the family table models of the screws:
> 
> Can we ask to DEDA? They might already have it.

but I've already done it ....

fiorisi, 2021-11-18:
We should define a common way to name fasteners.
I've drafted some guidelines, we have to approve during the next silo meeting: https://github.com/icub-tech-iit/cad-mechanics/wiki/Technical-specification-standards#icub-tech-standard---todo-approve-and-move-to-guidelines

marcellosavoldi, 2021-11-23:
I modified the family tables.
I have separated the ISO14579 from the ISO14580: now there are 2 separate family-tables.
the generics created are the following:

[image]

You can see a summary of the screws in the following file:
[viti-torx_study.xlsx](https://github.com/icub-tech-iit/cad-libraries/files/7586659/viti-torx_study.xlsx)

I have identified the names with the ISO as specified in the guidelines.
In the generics I have added the word "TORX" to facilitate the search without having to remember the single normative by type of screw.
the variants are identified as follows:

|                  variant                   |                                                      Description                                                              | note |
|:--------------------------------|:--------------------------------------------------------------------------------------------|:------|
| M3x04_TORX-ISO14583-A4  | M3 x 10 -ISO14583, TORX socket pan head machine screws fully threaded, A4          |          |
| M8x12_TORX-ISO14579-A2  | M8 x 12 - ISO14579, TORX socket head cap screws, fully threaded, A2                       |          |
| M3x10_TORX-ISO14580-A2  |  M3 x 10 , ISO14580, TORX socket head cap screws, low head, fully threaded, A2      |          |
| M3x10_TORX-ISO14581-A2    |  M3 x 10 -ISO14581, TORX socket countersunk flat head screws fully threaded, A2 |          |

The material is identify as stainless-steel 316

fiorisi, 2021-11-23:
Great @marcellosavoldi.

Few questions (some of which will become a standard to be used from now on family tables and ISO components) just to align with the @icub-tech-iit/silo-mech:
- `M3x04_TORX-ISO14583-A4` do we want to keep the material in the file name?
- `M3x04_TORX-ISO14583-A4` why `A4` steel? Maybe `A2` was not available?
- the name of the family tables contains also the "torx" word?

> In the generics I have added the word "TORX" to facilitate the search without having to remember the single normative by type of screw.

Do we really need to keep "TORX" in the instances name?

marcellosavoldi, 2021-11-23:
> * `M3x04_TORX-ISO14583-A4` do we want to keep the material in the file name?

If we want to create a "M.... _TORX-ISO14583-A2" screw?

> * the name of the family tables contains also the "torx" word?

Yes... I replicated it in the variants and in the description ... Do you prefert replace "TORX" with "Hexalobular" ?
> 
> > In the generics I have added the word "TORX" to facilitate the search without having to remember the single normative by type of screw.
> Do we really need to keep "TORX" in the instances name?

In a BOM it can be more easily noticed, even at a glance ....
I do not consider it indispensable, but useful!

> * `M3x04_TORX-ISO14583-A4` why `A4` steel? Maybe `A2` was not available?

I kept this question last because I note one thing.
I had chosen the screws with A4 material, because on the [Bossard website](https://www.bossard.com/eshop/global-en/screws-and-bolts-with-internal-drive/hexalobular-6-lobe-socket-pan-head-machine-screws-fully-threaded/p/20038/?category=01.100.100&index=4&q=%3Arelevance%3A74EC6A2E-3795-43DB-A517-12F3521785C5%3AHexalobular%2Bsocket%3AA87579DF-7F9C-4535-9A8D-87830D3D5D84%3AStainless%2Bsteel) the only M1.6 screws are of A4 steel ... and so I chose all the screws ISO14583 of that type.
Now I realized that instead the [Wurth](https://eshop.wuerth.it/Categorie-di-prodotti/ISO-14583-acciaio-inox-A2/31143511020119.cyid/3114.cgid/it/IT/EUR/?CatalogCategoryRef=31143511020119%40WuerthGroup-Wuerth-3114&SelectedFilterAttribut=%255B%255D) has the same screws (except the M1.6, which does not exist at all) in A2 ...Obviously [Bossard](https://www.bossard.com/eshop/global-en/screws-and-bolts-with-internal-drive/hexalobular-6-lobe-socket-pan-head-machine-screws-fully-threaded/p/5687/?category=01.100.100&index=3&q=%3Arelevance%3A74EC6A2E-3795-43DB-A517-12F3521785C5%3AHexalobular%2Bsocket%3AA87579DF-7F9C-4535-9A8D-87830D3D5D84%3AStainless%2Bsteel) also has A2 screws, but from M2 size.
Could it be worth considering all sizes M2 to M3 in A2 material and keeping M1.6 in A4 only?

Others opinions?

@fiorisi , @icub-tech-iit/silo-mech

fiorisi, 2021-11-24:
Thanks @marcellosavoldi.

Just few additional notes as discussed F2F:
- the alias of the instances should follow the [naming convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard)
- we keep the material in the instances name, this help in differentiating the screws. We decided to add also the resistance class (`70` for A2 and A4).
- Concerning the standard structure of the alias of a standard components (e.g. screws, pin, nut, etc), I agree with you we can have something like: `IDENTIFIER` `DIAMETER` `x` `LENGTH`-`COMMON NAME`-`ISOXXX`-`MATERIAL`. **If the @icub-tech-iit/silo-mech approves we have to write it in the guidelines.**
- Concerning the `COMMON NAME` word `TORX`, as you correctly pointed out, is a trademark. The correct name is Hexalobular, which is defined in the [ISO 10664](https://github.com/icub-tech-iit/cad-libraries/files/7594680/ISO-10664_.pdf). [Wikipedia says](https://en.wikipedia.org/wiki/Torx) that sometimes it is abbreviated as `6lobe`. The same `COMMON NAME` should be used in the generic and family table name.

> Could it be worth considering all sizes M2 to M3 in A2 material and keeping M1.6 in A4 only?

Fine for me.

marcellosavoldi, 2021-11-24:
I am making the agreed changes.
If you agree I would replace the word "TORX" with "6 LOBE" in the generic and variant filename and instead about the description I would use the name "Hexalobular screw".

summing up
|                  variant                   |                                                      Description                                                              | note |
|:--------------------------------|:--------------------------------------------------------------------------------------------|:------|
| M3x04-6LOBE-ISO14583-A4-70  | M3 x 10 , ISO14583, Hexalobular socket pan head machine screws fully threaded, A4-70          |          |
| M8x12-6LOBE-ISO14579-A2-70  | M8 x 12 , ISO14579, Hexalobular socket head cap screws, fully threaded, A2-70                       |          |
| M3x10-6LOBE-ISO14580-A2-70  |  M3 x 10 , ISO14580, Hexalobular socket head cap screws, low head, fully threaded, A2-70      |          |
| M3x10-6LOBE-ISO14581-A2-70    |  M3 x 10 , ISO14581, Hexalobular socket countersunk flat head screws fully threaded, A2-70 |          |

Generic : 

- 6lobe_iso14579.prt
- 6lobe_iso14580.prt 
- 6lobe_iso14581.prt
- 6lobe_iso14583.prt

... I hope the description isn't too long for CREO and Wingst ...
in other case I try to remove the "space" ....

@icub-tech-iit/silo-mech , @fiorisi

fiorisi, 2021-11-24:
> If you agree I would replace the word "TORX" with "6 LOBE" in the generic and variant filename and instead about the description I would use the name "Hexalobular screw".

On this I would like to have a feedback from the @icub-tech-iit/silo-mech.

Lawproto, 2021-11-24:
I personally prefer "TORX". I have no strong arguments against "6LOBE", though, I think I can live with it.

Whereas, I would rather **not** add the _fully threaded_ part in the description. Each fastener norm specifies a `b` length for each diameter. If `L < b`, the screw is fully threaded, otherwise `b` is the threaded length. There is no ambiguity to be evaded with further description.
[image]

marcellosavoldi, 2021-11-24:
> I personally prefer "TORX". I have no strong arguments against "6LOBE", though, I think I can live with it.
>
> Whereas, I would rather **not** add the _fully threaded_ part in the description. Each [...]

Actually I also prefer TORX, but Luca is right .... TORX is a registered name to define a hexalobe imprint ... it would be like calling Lorenzo the whole male gender of humanoid.
about the wording "full thread" I could agree with you, but I'm not sure that there are codes to differentiate the two types of screws (for example in screws with hexagon socket there are 2 different codes depending on whether they are fully threaded or partially threaded).
could we possibly use a pseudonym like FT or PT?
In the Bossard and Wurth web-site there are only fully threaded ... so I don't know ...
I am confused, I leave the decision to you.

@Lawproto 
@fiorisi 
@salvi-mattia 
@Mick3Lozzo 
@Elio75 
@divyashah 
@Andrea-Mor

Lawproto, 2021-11-25:
> Actually I also prefer TORX, but Luca is right .... TORX is a registered name to define a hexalobe imprint ... it would be like calling Lorenzo the whole male gender of humanoid.

I could imagine the point being this. We probably are going to call the imprint `T5`, `T8`, `T10`anyway, so we are discussing about "angel sex". Do whatever, I am fine with it.

> about the wording "full thread" I could agree with you, but I'm not sure that there are codes to differentiate the two types of screws (for example in screws with hexagon socket there are 2 different codes depending on whether they are fully threaded or partially threaded).

This surprises me, can you make a proper example?

Lawproto, 2021-11-25:
**Update:** I explored with @Andrea-Mor the fully threaded/partially threaded thing. I have to admit that things are not quite the way I figured them to be.
I retire my objection on this topic.

marcellosavoldi, 2021-11-26:
in any case there is another problem ...
I needed to reduce the digits of the descriptions ... by at least 4/5 letters ...
I'm working on it...

marcellosavoldi, 2021-11-26:
I need to reduce the description...
I found following tabs about Common Fastener Related Abbreviations ... what do you think?
https://www.aftfasteners.com/common-fastener-related-abbreviations/
https://www.boltdepot.com/fastener-information/Abbreviations.aspx

If I reduce the written "fully threaded" with " F " or "f.t."  I can save sufficient space...

the longer description is:
`M2.5x10 , ISO14581, Hexalobular socket countersunk flat head screw fully threaded, A2-70`
this is the maximum acepted from CREO:
`M2.5x10 , ISO14581, Hexalobular socket countersunk flat head screw fully threade`

could be:
`M2.5x10 , ISO14581, Hexalobular socket countersunk flat head screw F, A2-70`
or
`M2.5x10 , ISO14581, Hexalobular socket countersunk flat head screw f.t. , A2-70`

marcellosavoldi, 2021-12-10:
final decision on the description:
M3x10 , ISO14583, 6LOBE socket pan head machine screw, fully threaded, A2-70
M3x10 , ISO14580, 6LOBE socket low head cap screw, fully threaded, A2-70
M8x40 , ISO14579, 6LOBE socket head cap screw, fully threaded, A2-70
M10x30 , ISO14581, 6LOBE socket countersunk flat head screw, f. threaded, A2-70

final decision on the generic
6LOBE-ISOxxxx.prt

final decision on the Variants
M2x06-6LOBE-ISO14583-A2-70
M3x12-6LOBE-ISO14580-A2-70
M8x55-6LOBE-ISO14579-A2-70
M6x12-6LOBE-ISO14581-A2-70

marcellosavoldi, 2021-12-10:
I added the right family table in cad-libraries:

[image]

@icub-tech-iit/silo-mech , can you check it?

You can find the file with all screws in the following document:
https://istitutoitalianotecnologia.sharepoint.com/:x:/r/sites/icub-facility/_layouts/15/Doc.aspx?sourcedoc=%7B44369799-52C3-43AA-A7EC-65FA62EC54DA%7D&file=viti-torx_study_R1.xlsx&action=default&mobileredirect=true

marcellosavoldi, 2021-12-10:
What is the fastest way to insert the screws in the table below to WIngst:
https://istitutoitalianotecnologia.sharepoint.com/:x:/r/sites/icub-facility/_layouts/15/Doc.aspx?sourcedoc=%7B44369799-52C3-43AA-A7EC-65FA62EC54DA%7D&file=viti-torx_study_R1.xlsx&action=default&mobileredirect=true

@icub-tech-iit/silo-mech , @Fabrizio69 

see only the first TAB "Full-Torx-List"...
I also split the screws among all mechanics to start normalizing when the files are ok for all.
wait for my GO before starting normalization!!!

fiorisi, 2021-12-14:
Great @marcellosavoldi.

Please set the material following the [guidelines](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts) and the correct [technical standard](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#technical-standards).

[image]

marcellosavoldi, 2021-12-16:
I'd like to define the material used for the A2-70 types of fasteners instead of setting the densities by hand.
According to the following tables you can see the difference between the various ones types of screws: [^1]

[image]

Based on the class of the screw we can define the following materials: [^1]
[image]
[image]

reading various information on the net, following what I found [HERE](https://www.inoxforniture.it/info-tecniche/acciai-inossidabili-per-bulloneria/), I would define the A2-70 type as a 304 stainless steel [^2]

What do you think @fiorisi , can we use the following material?
[image]

[^1]: https://www.vitieutensili.com/download/estrapolato.pdf
[^2]: [image]

fiorisi, 2021-12-16:
Nice work @marcellosavoldi. I try to recap the main points, then I will copy this section in our [Technical-specification-standards](https://github.com/icub-tech-iit/cad-mechanics/wiki/Technical-specification-standards).

The mechanical properties of corrosion-resistant stainless steel fasteners are defined in the [ISO 3506](https://www.iso.org/standard/70045.html). 

More in detail, the naming convention of the material is composed as follows:
`LETTER` `NUMBER_1` `-` `NUMBER_2`

Concerning the `LETTER` we have 4 different cases:
- `A` for Austenitic steel
- `C` for Martensitic steel
- `D` for Duplex steel
- `F` for Ferritic steel

Concerning the `NUMBER_1` we have different chemical compositions. For Austenitic we have:
- `1`, `2`, `3`, `4`, `5` and `8` with different chemical composition and corrosion resistances.

Concerning the `NUMBER_2` we have different treatments depending on the steel used. For Austenitic we have:
- `50` for soft state (annealed)
- `70` for work hardening state
- `80` for high strength state
- `100` for high strength state

For our default screws material `A2` there are 3 different materials that can be used to manufacture the screw (usually depends on manufacturer if not specified):
- `AISI 304`
- `AISI 304 L`
- `AISI 305`

As a reference I add the datasheet:
- [F_021_EN.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7729146/F_021_EN.pdf)

fiorisi, 2021-12-16:
To conclude, I think that the best option is to create a material that specifies the class. 
As an example we can name it: `ISO3506_A2-70`. If you agree I modify the guidelines 

It is important to add the correct properties (add through pull request).

marcellosavoldi, 2021-12-17:
I checked the original materials that PTC offers within Creo.
I noticed that there are generic, but well-compiled materials
I could make a copy of this:
[image]

@fiorisi

Lawproto, 2021-12-17:
If it possible, I would put the generic material into a subfolder of our material database to not have them mixed up with the specific ones.
On a side note, I found that our library lacks copper, I ask you to add that too in case you proceed, @marcellosavoldi

marcellosavoldi, 2021-12-21:
I'm honestly not a fan of subfolders, but I can live with it

feedback from @fiorisi ?

fiorisi, 2021-12-21:
> I could make a copy of this:

I agree but the parameters shall be updated with the ones for the steel grade that we selected.
[image]

> If it possible, I would put the generic material into a subfolder of our material database to not have them mixed up with the specific ones.

For the time being I would also keep all the material files in the same directory.

Lawproto, 2022-01-04:
I report a somewhat abnormal behaviour concerning at least the library components `ISO 14580` and `ISO 14583`, but I suspect this may affect other new torx-related parts as well.
When I do any modification in an assembly which contains the overmentioned parts and save, I get a new Creo version of the part as well, **even though its model was not touched at all**.

[image]

This never happened with the fasteners I used until now. Also, please notice that the library component `iso2338_b` is used in the assembly too but does not get a new version after saving. This should be the default behaviour, in my opinion.

marcellosavoldi, 2022-01-07:
I will check the models created. However, it is quite common for Creo to save parts that have not been modified.
We never understood why, but it happens often.
[image]

marcellosavoldi, 2022-01-10:
[I'm waiting a pull request](https://github.com/icub-tech-iit/cad-libraries/pull/86)

MrAndrea, 2022-01-10:
@fiorisi do you have some feedback concerning the PR [mentioned above](https://github.com/icub-tech-iit/cad-libraries/issues/74#issuecomment-1008719579)?

Fabrizio69, 2022-01-10:
> What is the fastest way to insert the screws in the table below to WIngst: [...]

@marcellosavoldi the fastest way to insert screws in WingGST is to use mechanical CAD to generate a BOM to import, in order to create the new WinGST codes. After creating these new codes, the manufacturer and supplier information must be entered manually.

Lawproto, 2022-01-12:
So, @marcellosavoldi, am I free to start my share of WinGST insertions for these screws.

marcellosavoldi, 2022-01-12:
> So, @marcellosavoldi, am I free to start my share of WinGST insertions for these screws?

@Lawproto , Follow the informations on this new issue:
https://github.com/icub-tech-iit/cad-mechanics/issues/185

Lawproto, 2022-01-13:
@marcellosavoldi I noticed that some screw families have a conical head. Here is `ISO 14580` as an example.

[image]

While I see the aestethic touch the conical head wants to convey, it makes it quite difficult to measure the diameter size while working in Creo.

I am suggesting we transform it into a straight, cylindrical surface.

marcellosavoldi, 2022-01-13:
... Personally I prefer that the screws reflect reality as much as possible.

[image]

Lawproto, 2022-01-13:
I like your picture @marcellosavoldi.

[image]

Can we make is so that this `d_k`, which is a useful dimension, is easy to measure from the 3D model?

fiorisi, 2022-01-13:
@marcellosavoldi where did you take this documentation?

Just as a reference I've added the ISO standard in the [dedicated section](https://github.com/icub-tech-iit/cad-mechanics/wiki/Technical-specification-standards#hexalobular-screw---torx) (feel free to add also the others). 
[image]

Section 5 of the ISO also defines the designation.
[image]

marcellosavoldi, 2022-01-13:
you can find the dimensions in the wiki:
https://github.com/icub-tech-iit/cad-libraries/wiki/Hexalobular-screws-dimensions

marcellosavoldi, 2022-01-26:
I created a small cylindrical area on the base of the screw to facilitate the measurement.

[image]

I think that now all guys are happy and we can close the issue again...

closed 2022-01-26 by marcellosavoldi