mesh-iit/cad-libraries #75: [ADAFRUIT] - [1752] - [Stereo Audio Amplifier]

Mick3Lozzo, 2021-11-08:
| Description | 20W Stereo Audio Amplifier - MAX9744 |
| --- | --- |
| Datasheet component |  [adafruit-20w-stereo-audio-amplifier-class-d-max9744.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7499072/adafruit-20w-stereo-audio-amplifier-class-d-max9744.pdf), [MAX9744.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7499082/MAX9744.pdf) |
| Project | ergoCub, R1 |
| Alias Creo | ADAFRUIT_MAX9744 |
| IIT-Code | XXXXX |
| Applicant | Michele Gesino, @Mick3Lozzo  |
| Supervisor | Ugo Pattacini, @pattacini  |
| Maintainer | Marcello Savoldi, @marcellosavoldi |
| Coworkers | @icub-tech-iit/silo-mech |

### Attachment:

### Notes:

### Checklist:

Complete the checklist in order to proceed with the request.

*   [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
*   [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
*   [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
*   [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
*   [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
*   [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Mick3Lozzo, 2021-11-08:
[image]

Weight (found on the web): 11 g.  
  
@MrAndrea as I understood [here](https://github.com/icub-tech-iit/tickets/issues/243#issuecomment-812490314), this component has been ordered without adding it to WinGST database so it doesn't still have an IIT code, right?

MrAndrea, 2021-11-09:
@Mick3Lozzo I added just now on Wingst the Adafruit 1752 electronic board: iitcode **15585**.
If it is possible, I suggest to change the Creo alias from ADAFRUIT_MAX9744 to ADAFRUIT_1752.

fiorisi, 2021-12-06:
@Mick3Lozzo @MrAndrea which is the board code? MAX9744 or ADAFRUIT_1752?

fiorisi, 2021-12-07:
> @fiorisi the manufacturer code is Adafruit 1752.
MAX9744 is the chip soldered on the board.

Thanks @MrAndrea.

I think that we can close the issue @Mick3Lozzo.

closed 2021-12-07 by fiorisi