mesh-iit/cad-libraries #77: [Visaton] - [EX 30 S] - [speaker]

Mick3Lozzo, 2021-11-09:
| Description          | Speaker 10W, 8 Ohm, 25-70 C |
|:---------------------|:---------------------------------------|
| Datasheet component  [EX_30_S-2322537.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7503976/EX_30_S-2322537.pdf)|
| Project              | ergoCub    |
| Alias Creo           | VISATON_EX-30-S-8     |
| IIT-Code             | XXXXX     |
| Applicant            | Michele Gesino, @Mick3Lozzo   |
| Supervisor           | Ugo Pattacini, @pattaci    |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:
Weight: 40 g 

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Mick3Lozzo, 2021-11-09:
[image]

Checked and corrected mass.  
  
@MrAndrea still we don't have bought this item, so it doesn't have an IIT code, right?

marcellosavoldi, 2021-11-09:
@Mick3Lozzo , who buys this item ( mech or eltr team) ?
if it has to appear in Wingst's mechanical BOM, you have to put another parameter, because "eltr" is not read by Wingst ...

Mick3Lozzo, 2021-11-09:
@marcellosavoldi  saw another two Visaton speakers in our CAD repository, and they were both classified as  `ELTR`.

marcellosavoldi, 2021-11-09:
OK,
then we leave ELTR.
I just wanted to be sure that there was no need for this object to appear in the mechanical BOM.
In doing so, the electronics take care of it.

@Mick3Lozzo , @icub-tech-iit/silo-electr , @MrAndrea

MrAndrea, 2021-11-10:
Sorry but we can't continue with this speaker for ergoCub's head, look at [this comment](https://github.com/icub-tech-iit/ergocub-design-head/issues/50#issuecomment-964900858).

fiorisi, 2021-11-11:
@marcellosavoldi if the CAD model is correct probably we can keep it anyway in the library.
In this case feel free to close the issue.

marcellosavoldi, 2021-11-11:
I agree with Luca,
@Mick3Lozzo ,If it is already done, we leave it in the bookstore anyway, are you agree?
Anyway, @MrAndrea , can you confirm that it is not something that has to come out in mechanical BOM?

MrAndrea, 2021-11-11:
Yes @marcellosavoldi, the speaker will be in the electronics BOM.

fiorisi, 2021-11-11:
@marcellosavoldi if the component is ready we can close the issue.

marcellosavoldi, 2021-11-11:
> @marcellosavoldi if the component is ready we can close the issue.

... I leave the honor to @Mick3Lozzo