mesh-iit/cad-libraries #79: WIKI-PRELIMINARY _ List standard commercial parts

marcellosavoldi, 2021-11-15:
preliminary sketch that will move to WIKI

marcellosavoldi, 2021-11-15:
## Hexagon socket cap head
[image]
|   UNI  |   ISO  |  DIN  | file CAD        | Notes             |
|:------:|:------:|:------|:----------------|:------------------|
|  5931  |  4762  |  912  |  uni5931.prt    |                   |

Dimensions: M3, M4, M5, M6, M8, M10, M12
length : all standard length ( as possible fully tread )
material : Stainless steel
type and class : A2-70

## Hexagon socket countersink head
[image]
|   UNI  |   ISO      |    DIN  | file CAD          | Notes                 |
|:-------:|:---------:|:--------|:----------------|:------------------|
|  5933  |   10642  |  7991  |  uni5933.prt    |                           |

Dimensions: M3, M4, M5, M6, M8, M10, M12
length : all standard length
material : Stainless steel
type and class : A2-70

## Pozidrive cross recessed pan head
[image]
|   UNI  |   ISO      |    DIN    |     file CAD          | Notes                 |
|:-------:|:---------:|:---------|:------------------|:------------------|
|  7686  |   7045   |  7985A   |  iso7045_z.prt    |                           |

Dimensions: M1.6, M2, M2.5, M3
length : all standard length
material : Stainless steel
type and class : A2-70

## Philips cross recessed flat head
[image]
|   UNI  |   ISO      |    DIN    |     file CAD          | Notes                 |
|:-------:|:---------:|:---------|:------------------|:------------------|
|  7688  |   7046   |   965A   |  iso7046_h.prt    |                           |

Dimensions: M1.6, M2, M2.5, M3
length : all standard length
material : Stainless steel
type and class : A2-70

marcellosavoldi, 2021-11-18:
## Hexagon nuts
[image]
|   UNI  |    ISO  |  DIN  | file CAD          | Notes             |
|:-------:|:-------:|:------|:----------------|:------------------|
|  5588  |  4032  |  934  |  iso4032.prt    |                   |

Dimensions: M1.6, M2, M2.5, M3, M4, M5, M6, M8, M10, M12, M16, M20
material : Stainless steel

Dimensions: M2.5, M3, M4, M5, M6
material : Nylon

fiorisi, 2021-12-06:
@marcellosavoldi I close this issue.
We already have the wiki sections to edit.

closed 2021-12-06 by fiorisi

marcellosavoldi, 2022-01-10:
I need this issue to download the images and copy the link on the wiki.... it's the faster way!

marcellosavoldi, 2022-01-10:
[image]
[image]
[image]
[image]