mesh-iit/cad-libraries #81: [Harmonic-Drive] - [CPL-20A-160-2A and  CSD-20-100-2A-GR] - [Harmonic Drive Component set]

marcellosavoldi, 2021-12-13:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [[CSD data](https://harmonicdrive.de/it/prodotto/set-componibili/csd-2a) and [CPL data](https://harmonicdrive.de/it/prodotto/set-componibili/cpl-2a) ]                  |
| Project              | [Ergocub joint]     |
| Alias Creo           | [see notes]                         |
| IIT-Code             | [TBD]                         |
| Applicant            | Marcello Savoldi, @marcellosavoldi                 |
| Supervisor           | Luca FIorio, @lauracavaliere                |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

Quotation ( could be good for the normalization ):
[IIT-06-12-2021.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7704766/IIT-06-12-2021.pdf)
[IIT-06-12-2021_bis.pdf](https://github.com/icub-tech-iit/cad-libraries/files/7704770/IIT-06-12-2021_bis.pdf)

### Notes:
|                 ALIAS                 |                                                                  Description                                                |  Type   |   WAIT   |
|:-----------------------------|:------------------------------------------------------------------------------------------|:---------|:--------|
| CPL-20A-160-2A_A           | CPL-20A-160-2A, Harmonic Drive Gearbox, size 20, ratio 160, Circular spline          | TRAS    |              |
| CPL-20A-160-2A_B           | CPL-20A-160-2A, Harmonic Drive Gearbox, size 20, ratio 160, wave generator        | COSM  |              |
| CPL-20A-160-2A_C           | CPL-20A-160-2A, Harmonic Drive Gearbox, size 20, ratio 160, flex spline                 |COSM   |              |
| CSD-20-100-2A-GR_A      | CSD-20-100-2A-GR, Harmonic Drive Gearbox, size 20, ratio 100, Circular spline      | TRAS    |              |
| CSD-20-100-2A-GR_B      | CSD-20-100-2A-GR, Harmonic Drive Gearbox, size 20, ratio 100, wave generator    | COSM  |              |
| CSD-20-100-2A-GR_C      | CSD-20-100-2A-GR, Harmonic Drive Gearbox, size 20, ratio 100, flex spline             | COSM  |              |

|                 ALIAS                 |   IIT_CODE   |
|:-----------------------------|:--------------|
| CPL-20A-160-2A_A           |                     |
| CSD-20-100-2A-GR_A      |                     |

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

marcellosavoldi, 2021-12-14:
CPL-20A-160-2A series done!

Mass assigned :
full mass gear = 0.14Kg
full volume = 19456.5 mm^3
circular = 8359.3 mm^3 , 0.06015Kg
Wave = 7384.7 mm^3 , 0.05314Kg
Flex = 3712.5, 0.02671Kg

license comm = ok

IIT_CODE = 15804

marcellosavoldi, 2021-12-14:
CSD-20-100-2A-GR series done!

Mass assigned :
full mass gear = 0.13Kg
full Volume = 19069.1 mm^3
circular = 9040.1 mm^3 , 0.06163Kg
Wave = 8263.1 mm^3 , 0.05633Kg
Flex = 1765.9 mm^3, 0.01204Kg

license comm = ok

IIT_CODE = 15805

fiorisi, 2021-12-14:
Great @marcellosavoldi. Just one point that I think we can discuss at the next silo meeting.
When a 3D model is available from the manufacturer, I would use it and not design the part.

marcellosavoldi, 2021-12-17:
done!
if there are no other requests I would say that it can be closed

fiorisi, 2021-12-17:
Ok.

closed 2021-12-17 by fiorisi