mesh-iit/cad-libraries #82: [manufacturer] - [part number] - [short description]

Elio75, 2021-12-15:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | [where will the component be used]     |
| Alias Creo           | [maker + code]                         |
| IIT-Code             | [if available]                         |
| Applicant            | Name Surname, @username                |
| Supervisor           | Name Surname, @username                |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

fiorisi, 2021-12-15:
Great @Elio75.
If you commit the file we can check it.

Elio75, 2021-12-15:
Done

@fiorisi

fiorisi, 2021-12-16:
@Elio75 can you also add the parameters?
[image]

Elio75, 2021-12-17:
Done

Elio75, 2021-12-20:
`Where did you take the mass of the board?`

[image]

dall`analisi di creo ho preso il volume e diviso per massa per calcolare la densità 

@fiorisi

fiorisi, 2021-12-21:
Great @Elio75.
Please always add the picture if the mass is not reported in the datasheet.

closed 2021-12-21 by fiorisi