mesh-iit/cad-libraries #83: [manufacturer] - [part number] - [short description]

Elio75, 2021-12-15:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | [where will the component be used]     |
| Alias Creo           | KONTRON_34009-0000-99-2_R11                         |
| IIT-Code             | 15625                        |
| Applicant            |  @Elio75                |
| Supervisor           |  @fiorisi               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

fiorisi, 2021-12-16:
@Elio75 can you also add the part parameters? I think that we can add a checkpoint also for this.

[image]

Elio75, 2021-12-20:
DONE

@fiorisi

closed 2021-12-21 by fiorisi