mesh-iit/cad-libraries #85: [manufacturer] - [part number] - [short description]

Elio75, 2021-12-15:
| Description          | Carrier Board for COM-Express Type10, Connecttech CCG020                              |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              |ergoCub     |
| Alias Creo           | CONNECTTECH_CCG020                      |
| IIT-Code             | 15301                        |
| Applicant            |  @Elio75                |
| Supervisor           |  @fiorisi               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:
[CTI-X-COM-Express-Type-10-mini (1).pdf](https://github.com/icub-tech-iit/cad-libraries/files/7719030/CTI-X-COM-Express-Type-10-mini.1.pdf)

[image]

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

fiorisi, 2021-12-16:
@Elio75 can you add the missing parameters (IIT code if available)?
[image]

The file is too heavy, there are many details that we don't need. Is there a lightweight version?
[image]

Elio75, 2021-12-17:
Done

Elio75, 2021-12-20:
`The file is too heavy, there are many details that we don't need. Is there a lightweight version?`

ho fatto uno Shrinkwrap

marcellosavoldi, 2021-12-21:
I would like to propose making a shrinkwrap for the complete board as surfaces and to solidify the PCB only.
ould this decrease the weights?
So I wish the board had the MATE_CONN as described about the individual components [here](https://github.com/icub-tech-iit/mechatronics/wiki/How-to-create-a-library-component-in-Creo-Parametric#mandatory-coordinate-system-for-plug-components), at least for the components used ...
What do you think?

@Elio75 
cc @fiorisi

fiorisi, 2021-12-23:
> I would like to propose making a shrinkwrap for the complete board as surfaces and to solidify the PCB only.
ould this decrease the weights?
So I wish the board had the MATE_CONN as described about the individual components here, at least for the components used ...
What do you think?

Can you do it for this board and attach the file here so we have a look at the final result?

marcellosavoldi, 2021-12-29:
[pcb_sw0001.prt.zip](https://github.com/icub-tech-iit/cad-libraries/files/7789324/pcb_sw0001.prt.zip)

@fiorisi , @Elio75

fiorisi, 2022-01-03:
Thanks @marcellosavoldi.
The size of the file is more or less the same though. 

I did a test by manually removing the small features, but still the size is 11 Mb
[test_delete_small_features.prt.1.zip](https://github.com/icub-tech-iit/cad-libraries/files/7801619/test_delete_small_features.prt.1.zip)

If we don't find a simple way I would keep the source file if the file is no greater than 20 Mb.

What do you think?

marcellosavoldi, 2022-01-03:
could we try to reduce the number of surfaces (in the connectors for example) before doing the shrinkwrap?

fiorisi, 2022-01-28:
@elio could you try to reduce the size of the file as suggested? Then please upload the file here.

Lawproto, 2026-01-16:
The alias exists in both WinGST and cad-libraries. All boxes are properly ticked. Closing.