mesh-iit/cad-libraries #87: Replace HD components CSD-17-100-2A (_CS / _WG / _FS) with CSD-17-100-2A-R (_A / _B / _C) on ergoCub

Elio75, 2022-01-10:
| Description   | Field                              |
|:--------------|:-----------------------------------|
| Category      | [MCAD]                        |
| Alias Creo    | [CSD-17-100-2A-R_ (A/B/C]                   |
| IIT-Code      | [15482]                     |
| Applicant     | Elio Massa, @Elio75            |
| Supervisor    | Luca Fiorio, @fiorisi            |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @icub-tech-iit/silo-mech           |

## Description and motivation of the changing request: <br>
The CSD-17-100-2A (_CS / _WG / _FS) version is replaced on the ergoCub CAD because it is obsolete

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Checked that the associated [WINGST](http://wingst.icub.iit.local/) code is up-to-date and that the manufacturer and at least one supplier are defined.

Elio75, 2022-01-10:
Each HD consists of 3 parts

Old version
| Alias | Cod. WGST |  
|:---:|:---:|
| CSD-17-100-2A_CS| 7062 |
| CSD-17-100-2A_WG|
 |CSD-17-100-2A_FS|

New version
| Alias | Cod. WGST |  
|:---:|:---:|
|CSD-17-100-2A-R|  15482| 
 
Composed of

| Alias | Cod. WGST |  
|:---:|:---:|
|CSD-17-100-2A-R_A|15483|
|CSD-17-100-2A-R_B||
|CSD-17-100-2A-R_C||

Elio75, 2022-01-10:
List of assemblies where the HD components to be replaced are located

**ECUB_004_A_001_UPPERBODY**
ECUB_004_A_002 (Right arm )
CSD-17-100-2A  (_CS / _WG / _FS)
| Assembly | Alias | Component | Quantity | 
|:---:|:---:|:---|:---:|
| ECUB_004_G_020| CSD-17-100-2A_WG |  WG | 1| 
| ECUB_004_G_021| CSD-17-100-2A_FS |  FS | 1| 
| ECUB_004_G_021| CSD-17-100-2A_CS |  CS| 1| 
| ECUB_004_G_034| CSD-17-100-2A_CS |  CS| 1| 
| ECUB_004_G_035| CSD-17-100-2A_FS |  FS| 1|
 | ECUB_004_G_036| CSD-17-100-2A_WG |  WG| 1|

ECUB_004_A_003 (Left Arm)
CSD-17-100-2A (_CS / _WG / _FS)
| Assembly | Alias | Component | Quantity | 
|:---:|:---:|:---|:---:|
| ECUB_004_G_020| CSD-17-100-2A_WG |  WG | 1| 
| ECUB_004_G_021| CSD-17-100-2A_FS |  FS | 1| 
| ECUB_004_G_021| CSD-17-100-2A_CS |  CS| 1| 
| ECUB_004_G_034| CSD-17-100-2A_CS |  CS| 1| 
| ECUB_004_G_035| CSD-17-100-2A_FS |  FS| 1|
 | ECUB_004_G_036| CSD-17-100-2A_WG |  WG| 1|

@maggia80 @Fabrizio69

Elio75, 2022-01-10:
List of assemblies where the HD components to be replaced are located

**ECUB_003_A_001_LOWERBODY**
ICUB3_018_G_018 (Left foot)
CSD-17-100-2A (_CS / _WG / _FS)

Elio75, 2022-01-10:
You could do as in issue # 54, that the components to be replaced are made a sketch with the writing Obsolete, in order to be displayed and then replaced.

@icub-tech-iit/silo-mech

fiorisi, 2022-06-22:
@Elio75 can we close this issue?

closed 2022-06-22 by Elio75