mesh-iit/cad-libraries #89: ECAD - [AEDR-9820] - [IIT-CODE] - [Reflective Incremental Encoders]

marcellosavoldi, 2022-01-21:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | see [HERE](https://docs.broadcom.com/doc/AEDR-9830-3-Channels-Reflective-Incremental-Encoders-Output-318-LPI)              |       |
| Manufacturer info  | [Broadcom + AEDR-9830] * |       |
| Alias Creo         | [AEDR-9830] \**          |       |
| MCAD Contact       | Marcello Savoldi @marcellosavoldi                |       |
| ECAD Contact       | Maurizio Borgagni @maurizbo               |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)
[image]

###  Attachments (optional)
[image]
[image]

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

marcellosavoldi, 2022-09-21:
done!

closed 2022-09-21 by marcellosavoldi

MrAndrea, 2022-10-11:
Title changed with the right model AEDR-9830 and iitcode 16143.