mesh-iit/cad-libraries #90: [manufacturer] - [part number] - [short description]

Andrea-Mor, 2022-02-10:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | [where will the component be used]     |
| Alias Creo           | sauter_fh500                         |
| IIT-Code             | [if available]                         |
| Applicant            | Andrea Mor @Andrea-Mor                |
| Supervisor           | Luca Fiorio @fiorisi                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Andrea-Mor, 2022-02-10:
## Component mass

Sauter FH500 [datasheet](https://dok.kern-sohn.com/manuals/files/Italian/fh-ba-it-2020.pdf)

[image]

[image]
[image]

6.3 N = 0,642 kg ✔️

Andrea-Mor, 2022-02-10:
@fiorisi I ask your help for the last three points of the first comment

> Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.

I think this is not my case.

>  Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

do I have to convert it?

>  Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

again, this is not our case?

Andrea-Mor, 2022-02-16:
Design of the Dynamometer changed. An error was fixed

Andrea-Mor, 2022-02-23:
@fiorisi do I have to do something else?

I've converted the prt file with EDU-COM liecnse and I've checked that there's no special case for this component. 
As long as we don't have to create a IIT code in Wingst, it's all done!

fiorisi, 2022-02-23:
Thanks @Andrea-Mor. I think that we can close the issue.

closed 2022-02-23 by fiorisi