mesh-iit/cad-libraries #93: [TQ] - [ILM series] - [motor for ergoCub joints]

marcellosavoldi, 2022-03-29:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [[TQ link](https://www.tq-group.com/en/products/tq-robodrive/servo-kits/) ]                  |
| Project              | [ergoCub]     |
| Alias Creo           | [TQ_ILM... see table below]                         |
| IIT-Code             | [see table below]                         |
| Applicant            | Marcello Savoldi, @marcellosavoldi  and Divya Shah, @divyashah                |
| Supervisor           | andrea Mura, @MrAndrea                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

divyashah, 2022-04-06:
Since we have shortlisted the types of chosen motors, below is the updated table with the relevant parameters and the status of them: 

## Parts to do:

|  Motor   |  STATUS   |    Alias              | IIT CODE |   Description                   |
|:--------:|:---------:|:---------------------:|:--------:|:-------------------------------:|
| ILM50x08 | **to rename** | TQ_ILM50X08_SS_STATOR |   15577  | ILM50x08 star-serial withTemp 311773.0100 Stator |
|          | same name | TQ_ILM50X08_ROTOR     |   15567  | ILM50x08 176110.0100 Rotor |
| ILM50x14 |  renamed  | TQ_ILM50X14_SS_STATOR |   15578  | ILM50x14 star-serial withTemp 312111.0100 Stator |
|          | same name | TQ_ILM50X14_ROTOR     |   15570  | ILM50x14 286834.0101 Rotor |
| ILM70x10 | committed | TQ_ILM70X10_DS_STATOR |  **to add**  | ILM70x10 delta-serial withTemp 315235.0100 Stator |
|          | committed | TQ_ILM70X10_ROTOR     |  **to add**  | ILM70x10 176094.0100 Rotor                          |

MrAndrea, 2022-04-07:
I think now we have to estimate the number of the [motors above](https://github.com/icub-tech-iit/cad-libraries/issues/93#issuecomment-1090471427) in order to go with the purchase.

divyashah, 2022-04-07:
> I think now we have to estimate the number of the [motors above](https://github.com/icub-tech-iit/cad-libraries/issues/93#issuecomment-1090471427) in order to go with the purchase.

Yes @MrAndrea I created a table of motors/harmonic drives and their estimated quantities in https://github.com/icub-tech-iit/ergocub-design-joint/issues/69#issuecomment-1090524265.

divyashah, 2022-04-13:
All the motors from the shortlisted table above in https://github.com/icub-tech-iit/cad-libraries/issues/93#issuecomment-1090471427 have been added/renamed.  

Maybe you can close this issue ?? 
cc @marcellosavoldi

closed 2022-04-15 by divyashah