mesh-iit/cad-libraries #94: IGUS - GSM-0506-046 - sleeve bearing

Lawproto, 2022-04-04:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | https://www.igus.it/product?artNr=GSM-0506-046 |
| Project              | ergoCub lower arm (Mk2) |
| Alias Creo           | IGUS_GSM-0506-046          |
| IIT-Code             | [to be updated]                         |
| Applicant            | Lorenzo Protopapa, @Lawproto       |
| Supervisor           | Luca Fiorio, @fiorisi             |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Lawproto, 2022-04-04:
Checklist complete. Item added to `cad-libraries` and WinGST. Closing.

closed 2022-04-04 by Lawproto

fiorisi, 2022-04-04:
@Lawproto did you use the correct start part? 
https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components

Lawproto, 2022-04-04:
> @Lawproto did you use the correct start part? https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components

No, I simply assigned the Iglidur material in this case.

fiorisi, 2022-04-04:
Ok, then simply remove these parameters from the part. 
[image]

Lawproto, 2022-04-04:
I see your point, @fiorisi. Fixed that.