mesh-iit/cad-libraries #97: [Technogym] - [Urethane-plates] - [short description]

marcellosavoldi, 2022-05-12:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [ [wbsite ](https://www.technogym.com/it/urethane-plate-set.html)]                  |
| Project              | setup and others    |
| Alias Creo           | [Technogym]                         |
| IIT-Code             | 16241,16242, 16243, 16244    |
| Applicant            | Marcello Savoldi, @marcellosavoldi               |
| Supervisor           | Luca Fiorio, @fiorisi                |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:
[image]
[image]

|                        |     Foro asta    |     D Max     |     H                |   |
|------------------------|------------------|---------------|----------------------|---|
|     Peso da 1.25 Kg    |     50 mm        |     166 mm    |     21.5 ± 0.5 mm    |   |
|     Peso da 2.5 Kg     |     50 mm        |     198 mm    |     27.5 ± 0.5 mm    |   |
|     Peso da 5 Kg       |     50 mm        |     240 mm    |     38 ± 0.5 mm      |   |
|     Peso da 10 Kg      |     50 mm        |     320 mm    |     40 ± 0.5 mm      |   |
|     Peso da 20 Kg      |             |               |                      |   |
|                        |                  |               |                      |   |

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

marcellosavoldi, 2022-05-13:
I created the family table technogym_training_plate.prt
[image]

but I've some problems to define the different density for all plates...
Who can help me?

@icub-tech-iit/silo-mech

fiorisi, 2022-05-13:
@marcellosavoldi in this case I think that we have to rely on the `Alternate mass properties`. 
See for more details: https://www.ptc.com/en/support/article/CS187047

If you can try this option, then we can discuss it and approve it at the next silo-meeting.

divyashah, 2022-05-13:
I had the similar problem of assigning mass properties for family tables for the issue of Seeger Rings (https://github.com/icub-tech-iit/cad-libraries/issues/56). We decided not use seeger rings at that time, so that issue got pushed into icebox. 

> If you can try this option, then we can discuss it and approve it at the next silo-meeting.

Now, I can try to help look into this and I think we already have this point for discussion in our list for the silo-mech meeting.  

@marcellosavoldi @fiorisi

divyashah, 2022-06-29:
Removed additional parameters from the file to be coherent with the start-part commercial. Closing the issue now.

closed 2022-06-29 by divyashah